bruhautomation/ESP-MQTT-JSON-Digital-LEDs #8: Memory when sending On command.

MichaelTranster, 2017-03-11:
When you toggle the "off" switch on HA and turn it back on. Is it possible to make the NodeMCU display the same effect that it was before the shut down? 
Currently it resets to solid everytime you toggle on and off. I've been looking through the arduino sketch and couldn't seem a way to implement it. Anyone else want it to operate this way? THANKS AGAIN FOR BEING AWESOME.

bruhautomation, 2017-04-01:
For sure! Sorry for my delay in getting back to you on this. Yep, I fixed in my recently merged code. Cheers!

closed 2017-04-01 by bruhautomation

MichaelTranster, 2017-04-02:
I can't seem to get the effects to work. There's not a way to select an effect via HA.

bruhautomation, 2017-04-02:
In order to use an effect, you have to turn the light on and the effect
drop down should be in the card. Or, you can use the Services tool with the
commands I listed in the readme file. Cheers!

MichaelTranster, 2017-04-02:
When I turn on the Light, no effect pops up. But when I click on the empty space above the work "Brightness" on the card, the word Effect and the arrow pops up. Clicking it does nothing but the clicking animation.  I've tried different "spacing" of the effect section of the configuration.yaml file but still an inactive drop down menu.  Attached is the screen shoot of my card with the issue!
[image]

bruhautomation, 2017-04-02:
Sorry for the troubles. I'd make sure your LEDs are functioning correctly using a FastLED sample script like DemoReel100 -> https://github.com/FastLED/FastLED/tree/master/examples/DemoReel100.  If that isn't functioning correctly, then it maybe a wiring or most likely a power issue.  

As for the effect drop down, it's most likely an issue with your HA config. What version of HA are you running? Can you attach the relevant parts of your config for your light setup?

MichaelTranster, 2017-04-07:
I was running an older version of HA because when i attempted to update it, the script failed. Don't know why but i just grabbed another SD card and installed a fresh version of HA via the Hassbian image. Seted up MQTT and just put the MQTT LIGHT JSON component. The effects are now working. But now on the the effects freeze constantly. I plugged the NodeMCU into a PC to check the Serial Monitor. It looses connection to MQTT and tries to reconnects and sometimes it reconnects and sometimes it succeeds. 

I'm sorry for the bombardment of messages about this issue. I'm just documenting this to see if anyone would have the same issue! Maybe its just me, you obviously got it to work! This new revision is just not liking me! Thanks for all the inspiration.