bruhautomation/ESP-MQTT-JSON-Digital-LEDs #9: HA color picker issue on solid animation on version Apr 2, 2017

bfqrst, 2017-04-07:
Hey guys,

first of all let me thank you for your work on the sketch. It works like a charm!

Here's an issue I'm struggling with: When you pick a color, the bulb icon will reflect the picked color correctly, while the actual color being sent to the strip is the mirrored one from the color picker. (Kinda hard to explain actually...) Some examples..

Solid animation:

- You pick blue
- Bulb icon color: blue
- Strip color: green

- You pick yellow
- Bulb icon color: yellow
- Strip color: purple

It picks the mirrored color if you look at that HA color picker... I hope you get the idea.
I changed the `#define COLOR_ORDER BRG` to `RGB` but that screwed up all the colors in all other animations. Sort of expected at this point... I also tried the color picker on a Yeelight strip and a Yeelight bulb as well as on a LIFX bulb. Works as expected. Colors are selected correctly.

Now the question is, is this a HA MQTT JSON Light [color picker] thing, or is it a ESP-MQTT-JSON-Digital-LED thing?

Can you guys maybe confirm this behavior?

Cheers,
Ralph

craig3050, 2017-04-07:
I had this exact same behaviour. It seems strange, as if you send colour and brightness values direct from HASS (via an automation script) it displays the correct colour, it just seems the colour chooser is not aligned. Somehow it seems if you could move it to the right by about an inch it would be perfect...

And yes, agreed, awesome work on the sketch. Saved me so many hours in working out how to make the effects!

bfqrst, 2017-04-07:
Holy smoke, I really wasn't sure somebody would understand my explanation... :-)

> It seems strange, as if you send colour and brightness values direct from HASS (via an automation script) it displays the correct colour

I had not tried this yet, it certainly was on my agenda but it doesn't surprise me that if you pass along the RGB values it would work the way it's intended to...

I remember I had this sort of behavior when I got my first LIFXs, but appearantly this got sorted out by now. Let's just see what @bruhautomation has to say about this and maybe we'll need to raise this issue some place else.

sesametoy, 2017-04-09:
Hi:
i have same issue in my led strip.  after i changed rgb order in .ino file problem solved. 

#define COLOR_ORDER GRB (use to be BRG)

bfqrst, 2017-04-09:
Thanks @sesametoy , indeed that did the trick for me too! Now that I look at it, it makes perfect sense. I actually WAS talking about colors being mirrored, but somehow failed to see that `BRG` and `GRB` IS exactly that mirrored state... 

Well anyway, thanks.
Ralph

bruhautomation, 2017-04-09:
Sorry for my delay getting back to you @bfqrst. @sesametoy is all over this one. You'll need to change the lines to match the color order of your LEDs.  Pumped you guys got it resolved. Let me know if I can help with anything. I'll try and do better about staying on top of messages after I get my sensor node video up soon. Cheers!

closed 2017-04-09 by bruhautomation