bruhautomation/ESP-MQTT-JSON-Digital-LEDs #11: cant set led effect in automation.

sesametoy, 2017-04-09:
cant set led effect in automation.  it works fine if without any effect defined. 
code blow 

- alias: switch ledon
  trigger:
    platform: event
    event_type: click
    event_data:
      entity_id: binary_sensor.switch_158d000XXXXXXX
      click_type: single
  action:
    service: light.turn_on
    entity_id: light.ledstrip1
    data_template:
      brightness: "250"
      color_name: "white"
      transition: "1"
      effect: "solid" **only works fine after comment this line. no any log pop up**

bruhautomation, 2017-04-09:
I can't see your formatting here, but should work without an issue.  Try to play around in the service tool to see if the light.turn_on service is working okay. In my setup I can go to... 

Service Tools -> Light.Turn_On and then run this JSON command and it works 
```
{"entity_id":"light.nightstand",
"color_name":"white",
"brightness":255,
"transition":255,
"effect":"solid"
}
```
Can you confirm the same on your end?

sesametoy, 2017-04-10:
nop. 
only led can turn on without "effect" parameters
no idea what's happend

in ino file, i only changed RGB order and topic name.

sesametoy, 2017-04-10:
oh  another thing
i turnoff MQTT persistent. not sure if it will effect to issue.

g3n35i5, 2017-04-10:
I have the same issue here. When i subscribe to the topic, the service data
```
{"entity_id":"light.desk_wall",
"color_name":"white",
"brightness":255,
"transition":255,
"effect":"rainbow"
}
```
produces the following output:
```
Client mosqsub/[my device] received PUBLISH (d0, q0, r0, m0, 'g3n35i5/desk_wall/set', ... (94 bytes))
{"color": {"r": 255, "b": 255, "g": 255}, "brightness": 255, "state": "ON", "transition": 255}
```

As you can see, the effect tag is completely missing there. Do you have any idea what causes this error?

bruhautomation, 2017-04-11:
If you drop the transition line, can you call effects okay otherwise? Are the effects functioning as expected from the UI card?

bruhautomation, 2017-04-11:
That is very strange. When I call

bruhautomation, 2017-04-11:
That is very strange. When I call...

```
{"entity_id":"light.desk_wall",
"color_name":"white",
"brightness":255,
"transition":255,
"effect":"rainbow"
}
```

Using the mosquitto_sub command, I can see 
```
{"state":"ON","color":{"r":255,"g":255,"b":255},"brightness":255,"effect":"rainbow"}
```

Can you guys post your HA config?

sesametoy, 2017-04-12:
as i checked HomeAssistant github page.  seems there is a feature update in light mqtt_json. 
link as below
[https://github.com/home-assistant/home-assistant/pull/6941](url)
im not sure if its related to issue or not

sesametoy, 2017-04-12:
some updates, 
i upgrade HA to 0.42.2. and mqtt broker runs fine with "effect" .  but strange is cant work with "transittion"  (for light.trun_on service)
for light.turn_off service, it can only work with "transition" other parameters doesn't work at all.

g3n35i5, 2017-04-12:
Sorry for the late reply.
I'm currently struggeling with another problem, no matter how many LEDs i set in the ESPs code, only the first few LEDs are working. I'll see to fix the effect problem afterwards.

sisensee, 2017-04-14:
I currently don't have the ESP code deployed on any ESP's but I get a switch bounce, no color selections, and no effect choices in HASS. Any thoughts?

bruhautomation, 2017-04-17:
Hello. Sorry for the delayed response and the issues. I would highly recommend that you guys try this sample sketch first to make sure your LEDs are behaving as expected. https://github.com/FastLED/FastLED/blob/master/examples/DemoReel100/DemoReel100.ino

You'll need to change the relevant LED settings for your LEDs, but  it should work without any glitches if everything is setup correctly. 

@sesametoy, you use the service tool and call the exact same service settings I'm using and then post what is coming up in your MQTT broker?

g3n35i5, 2017-04-17:
Hello! Somehow now everything is working without changing anything. Im confused but it works.
The third effect i've tried crashed the program, i'll open an issue for it.
Thank you for your great support so far, your videos and tutorials really inspire me!

sesametoy, 2017-04-19:
this is what i input into service page 
```
{"entity_id":"light.ledstrip1",
"color_name":"white",
"brightness":255,
"transition":255,
"effect":"rainbow"
}
```

and this is what i got from mqtt broker:

```
Client mosqsub/19014-hassbian received PUBLISH (d0, q0, r0, m0, 'sesame/ledstrip1/set', ... (115 bytes))
{"color": {"g": 255, "b": 255, "r": 255}, "state": "ON", "transition": 255, "effect": "rainbow", "brightness": 255}

```
but still led doesn't light on, only if delete "transition": 255.

zuidwijk, 2017-04-21:
Try transition 0 (instant on)
Or try 5...10 (time fading from off till on in seconds).

As far as I know (from what I've seen in the code by testing) transition 255 means ~255 seconds until the transition is complete. 300 seconds is 5 minutes, 255 seconds will be 4:15 (approximately)

zuidwijk, 2017-04-21:
Maybe it is the max length? Sending 
`{"brightness": 255, "color": {"r": 255, "b": 255, "g": 255}, "effect": "rainbow" , "state": "ON", "transition": 255}`
the esp doesn't respond (no "Message arrived" on the terminal)

Sending:
`{"brightness": 255, "color": {"r": 255, "b": 255, "g": 255}, "effect": "rainbow" , "state": "ON", "transition": 255}`
`{"state": "ON"}`
Does work ;)

bruhautomation, 2017-04-23:
Sorry about the issues. Yeah, so the transition bit in this sketch is bit strange. When the LED is solid, transition is in seconds. When an effect is called, transition is used on this line `FastLED.delay(1000 / transitionTime);` which adds a delay in the rate at which the strip updates. That's the only way I know to "slow" an effect without hard coding some math. 

Can you guys confirm that you updated your PubSub libraries to support a MQTT packet size of 512? 
You can also do this by adding the line `#define MQTT_MAX_PACKET_SIZE = 512;` before setup.

zuidwijk, 2017-04-24:
I've added `#define MQTT_MAX_PACKET_SIZE = 512;` before setup, the command `{"brightness": 255, "color": {"r": 255, "b": 255, "g": 255}, "effect": "rainbow" , "state": "ON", "transition": 255}` doesn't arrive (no output at serial and mosquito)

zuidwijk, 2017-04-24:
Found this in a zipfile:
```
// MQTT_MAX_PACKET_SIZE : Maximum packet size
#ifndef MQTT_MAX_PACKET_SIZE
#define MQTT_MAX_PACKET_SIZE 128
#endif
```

bruhautomation, 2017-04-24:
Ah yeah, my bad. I think that line should be `#define MQTT_MAX_PACKET_SIZE 512` I think I added an extra equal. If you're not seeing the command come through in Mosquitto, then the issue is most likely with the PubSub library last night. Try dropping that equal sign and see if that does it.

g3n35i5, 2017-05-26:
@bruhautomation everything works fine by now, thank you for your work!

bruhautomation, 2017-06-02:
Nice! Pumped everything is working for you. Let me know if I can help with anything else. 

@zuidwijk @sesametoy feel free to reopen an issue if you're still having problems. Sorry about the slow response.

closed 2017-06-02 by bruhautomation