bruhautomation/ESP-MQTT-JSON-Digital-LEDs #12: ESP8266 loses MQTT connection when setting effect or animation speed VER. 10 Apr 2017

LunaLens, 2017-04-12:
I am using the example HA config and unmodified code (save for topics and network info)
When I set any effect besides Solid, subsequent commands do not work properly. Selecting a different color does nothing and turning the lights off causes the ESP to to set RGB to 0 several times and then lose connection to the MQTT server.

The only thing that seems to work properly is the animation speed set command.
Ironically, changing the animation speed before setting an effect causes the ESP to  loop setting RGB to the same value some 1000 times (i/e)
Setting LEDs:
r: 255, g: 255, b: 255
Loop count: 82
Setting LEDs:
r: 255, g: 255, b: 255
Loop count: 83
Setting LEDs:
r: 255, g: 255, b: 255
Loop count: 84
Setting LEDs:
r: 255, g: 255, b: 255
Loop count: 85

eldoner, 2017-04-14:
I got it too. The loop is in the code. But why, and for what?

LunaLens, 2017-04-14:
Did some poking and the issue is definitely the code and not MQTT. Question now is would this happen if you had nothing connected to the MCU? I don't actually have my LEDs yet, but figured I'd prep everything else.

eldoner, 2017-04-14:
@LunaLens I have exactly the same situation. I will test it with LEDs in few days. But I don't think this is the problem. As far I know, LEDs don't provide any feedback? In code is loop:

` if (loopCount <= 1020)`

Why 1020? I have no idea. Also,  can you check issue "Infinite "Setting LED"? Thank you.

bruhautomation, 2017-04-23:
Sorry about the issue! I am unable to replicate this on my end. Can you post the MQTT messages that are being sent on your broker from HA to your ESP and vice-versa? 

If you run Mosquitto MQTT, you can use the command `mosquitto_sub -d -u yourMQTTusername -P yourMQTTpassword -t "#"` in the terminal window to see the messages.

eldoner, 2017-04-24:
@bruhautomation 

```
Message arrived [bruh/porch/set] {"state": "ON"}
őţ?ŕ˙?i�ţ?'L @state
Message arrived [bruh/porch/set] {"state": "ON", "effect": "solid"}
solid
Setting LEDs:
r: 140, g: 147, b: 47
Message arrived [bruh/porch/set] {"transition":61}
transition
Setting LEDs:
r: 140, g: 147, b: 47
Setting LEDs:
r: 140, g: 147, b: 47
Setting LEDs:
r: 140, g: 147, b: 47
Setting LEDs:
r: 140, g: 147, b: 47
Setting LEDs:
r: 140, g: 147, b: 47
Setting LEDs:
r: 140, g: 147, b: 47
Setting LEDs:
r: 140, g: 147, b: 47
Setting LEDs:
```
- turn on LEDS
- set effect to solid
- change the speed of animation

LunaLens, 2017-04-26:
@bruhautomation 

Okay, took me a bit to get back (accidentally fried my NodeMCU >.<)
This is what HA sends.
```Client mosqsub/4345-hassbian received PUBLISH (d0, q0, r0, m0, 'lens/garage/rgb/set', ... (15 bytes))
{"state": "ON"}
Client mosqsub/4345-hassbian received PUBLISH (d0, q0, r0, m0, 'lens/garage/rgb/set', ... (16 bytes))
{"transition":1}```

and this is what the NodeMCU sends back (starting with initial state)
```Client mosqsub/4344-hassbian received PUBLISH (d0, q0, r0, m0, 'lens/garage/rgb', ... (83 bytes))
{"state":"OFF","color":{"r":255,"g":255,"b":255},"brightness":255,"effect":"solid"}
Client mosqsub/4344-hassbian received PUBLISH (d0, q0, r0, m0, 'lens/garage/rgb', ... (82 bytes))
{"state":"ON","color":{"r":255,"g":255,"b":255},"brightness":255,"effect":"solid"}
Client mosqsub/4344-hassbian received PUBLISH (d0, q0, r0, m0, 'lens/garage/rgb', ... (82 bytes))
{"state":"ON","color":{"r":255,"g":255,"b":255},"brightness":255,"effect":"solid"}```

After that the ESP loses MQTT connection and stops responding.

bruhautomation, 2017-04-27:
Oi, sorry about the fried board. You'll get a collection of those after a while.. 🥇 

Sorry for my delay, I've attempted to recreate this in my setup, but I can't seem to be able to. Can you guys tell me again what is working and what isn't? It'd be great to see that what the Mosquitto output (or whatever broker you are using), too, as you describe the step that isn't working.

I know @LunaLens, you just sent that. I'm struggling to understand what you mean with the "loop is running". The MQTT dropping is odd and is likely linked to whatever garbage text is fouling the JSON commands. 

I wish I could be more help, but that's pretty limited without being able to recreate the problem. 😢 

Can you guys confirm the FastLED demo reel sketch is working okay? I've seen issue where hardware issues will foul the sketch and cause it to crash randomly.

LunaLens, 2017-04-27:
@bruhautomation 

Okay, to be more specific when I change the effect it appears to work fine and reports its state to HA properly, but after a while it drops the connection, reconnects and seems to set the LEDs to 0 without reporting this to HA. If I try to do anything after that it will continously print
Setting LEDs:
r: 0, g: 0, b: 0
to serial.

If I try to change the animation speed while the 'solid' effect is selected it'll go into that for loop mentioned. No matter what I do it eventually loses connection and stops trying.

There are two things that I'm going to try. First I'll use the updated .ino and second I want to try actually connecting the LEDs. Unfortunately ChinaPost doesn't seem to be enjoying my shipment and is taking its sweet time.
I can try thew new ino tomorrow, but unfortunately I can't actually connect RGB LEDs.
It'd probably best if I left this issue alone until then, hopefully not more than a week.

As for posting more info and printouts, well I've posted pretty much anything relevant to the issue, unless there's some other output I'm missing.

The loop I'm talking about was this. I think this is what's running before it starts printing garbage and losing connection
```      if (loopCount <= 1020) {
        lastLoop = now;

        redVal = calculateVal(stepR, redVal, loopCount);
        grnVal = calculateVal(stepG, grnVal, loopCount);
        bluVal = calculateVal(stepB, bluVal, loopCount);

        if (effectString == "solid") {
          setColor(redVal, grnVal, bluVal); // Write current values to LED pins
        }
        loopCount++;
      }

bruhautomation, 2017-06-02:
Okay, so I've been doing some testing on this. Can you guys remove these lines 23 and 24 from your code and tell me if that helps your stability at all? 

```
#define FASTLED_INTERRUPT_RETRY_COUNT 0 //makes animations smoother... supposedly... 
#define FASTLED_ALLOW_INTERRUPTS 0
```

I'm at 24 hours and no crashes yet. Sorry about taking forever to come up with something for this!

bruhautomation, 2017-06-03:
Seems like people are reporting that removing those two lines did the trick. Feel free to open another issue if you have more problems. Cheers!

closed 2017-06-03 by bruhautomation