bruhautomation/ESP-MQTT-JSON-Digital-LEDs #13: Juggle animation running on 240-ish to full brightness crashes the strip on version Apr 11, 2017

bfqrst, 2017-04-14:
Hey guys,

well the title says it all. I noticed that when you're on full brightness and you switch to `juggle` animation, the strip shuts down. Initially I thought this only happens at 255 brightness, but it starts somewhere around 240 I think... 

At first I thought this is a logic level shifter problem so I replaced the MOSFET shifter with a TI 74HCT125N, which is considered top notch... Problem persists.

Now while this has _too much LEDs for your power supply_ written all over it, I'm asking nonetheless... Can anybody confirm?

BTW: I have a 5V 10A power supply for 100 WS2812B LEDs, which I thought should be sufficient. 

Cheers
Ralph

eldoner, 2017-04-14:
Your power supply is fine. I have 300 LEDs and my esp crashes too when I switch to juggle.

bfqrst, 2017-04-14:
Thanks for confirming @eldoner . Could be a math operation for brightness that somehow ends up being > 255? Uncaught exception maybe?

eldoner, 2017-04-14:
@bfqrst mine crashes at ~150 brightness. But it is for 300 LEDs. The EXCCAUSE says: 

`Load or store to an unaligned address`

bfqrst, 2017-04-15:
@eldoner is your controller by any chance a Wemos D1 mini pro? These are packing 16M bytes(128M bit) of flash. I read somewhere that it is not so trivial to make use of all the space because for instance esptool can't write more than 4MB without tweaking... I intended to use MicroPython with these but failed to flash them because of the aforementioned problem... On a similar note in the Arduino IDE you select WeMos D1 R2 & mini with 4M flash. From where I'm standing, this is neither a WeMos "pro" nor 16M flash. Could be nothing though... I have some spare Adafruit Feather Huzzahs, maybe I should try those.

Cheers
Ralph

eldoner, 2017-04-15:
@bfqrst no,  I have nodemcu lolin. I set in the Arduino IDE 4M flash.

bfqrst, 2017-04-15:
Okay, here we go. It's not or not only connected to brightness, it also has to to with the intensity of a picked color. Try this:

1. Turn on strip
2. Select solid animation
3. Leave or set brightness to max
4. Pick a color close to the bottom (closer to black)
5. Set animation to juggle
6. No crash
7. Pick a color closer to the top (closer to white)
8. Crash!

Right?

eldoner, 2017-04-15:
yes, indeed. I have no idea what is goin here :D

bruhautomation, 2017-04-17:
Sorry about the issues on this. There is definitely some debugging to do in some of the effects. Thanks much for bringing this up. I can confirm I get the same behavior as @bfqrst following his steps. Very strange. I'm looking into that and will let you know what I find.

bfqrst, 2017-04-17:
No worries @bruhautomation . Thanks a lot.

Cheers
Ralph

g3n35i5, 2017-04-17:
Obviously no need to open a new issue, there it is :D
I dont know if this helps but here is the monitor output from the crash:
```
Message arrived [bruh/porch/set] {"effect": "juggle", "state": "ON"}
jugglejuggle

Exception (9):
epc1=0x4020a565 epc2=0x00000000 epc3=0x00000000 excvaddr=0x3ffef0ff depc=0x00000000

ctx: cont 
sp: 3fff0200 end: 3fff0450 offset: 01a0

>>>stack>>>
3fff03a0:  3ffef140 3ffeed20 3ffef1a0 40202f34  
3fff03b0:  3fffdad0 00000000 3ffeef50 402036d3  
3fff03c0:  82004384 24810033 00177f00 76000c7b  
3fff03d0:  07690004 00096000 52020c59 134b040f  
3fff03e0:  04135206 6202125b 156d0114 00187900  
3fff03f0:  0fc2ff11 ff04a6c0 e1ff21b1 40202d0e  
3fff0400:  0000006e 00000011 000000a0 00000003  
3fff0410:  00000002 00000000 ffffffff feefeffe  
3fff0420:  00000000 00000000 00000001 3ffef428  
3fff0430:  3fffdad0 00000000 3ffef421 40209fec  
3fff0440:  feefeffe feefeffe 3ffef430 40100718  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(1,6)

 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset

```

@bruhautomation thank you so much for your help!

bruhautomation, 2017-04-23:
Sorry about the issue! I just uploaded a new version of the code to resolve this. Turns out it's a bug from the beatsin16 function in the FastLED library.  Subtracting 1 from the NUM_LEDS value did the trick. Let me know if you have any more issues! cheers!

closed 2017-04-23 by bruhautomation

bfqrst, 2017-04-24:
Thanks @bruhautomation !