bruhautomation/ESP-MQTT-JSON-Digital-LEDs #15: Infinite "Setting LED"

eldoner, 2017-04-14:
When I set effect other than "solid" and I turn off leds I get output: 

Message arrived [bruh/porch/set] {"state": "ON"}
?ěôţ?Ĺ“ @ěôţ?Ŕ˙?i�ţ?SL @state
Message arrived [bruh/porch/set] {"effect": "lightning", "state": "ON"}
lightninglightning
Message arrived [bruh/porch/set] {"state": "OFF"}
? ˙?SL @state
Setting LEDs:
r: 0, g: 0, b: 0
Setting LEDs:
r: 0, g: 0, b: 0
Setting LEDs:
r: 0, g: 0, b: 0
Setting LEDs:
r: 0, g: 0, b: 0
Setting LEDs:
r: 0, g: 0, b: 0
Setting LEDs:
r: 0, g: 0, b: 0
Setting LEDs:
r: 0, g: 0, b: 0

And eps writes it until I turn on leds again. 
Also as you can see, effect name is doubled.

LunaLens, 2017-04-15:
Ah yes, I'm all to familiar with garbage output on the serial monitor. This is pretty common for me. For starters, I changed the baudrate down to 9600 (I couldn't get any output at all on 115200.)

My best guess is that it may be linked to the issue I submitted about changing effect/animation speed. Once I change the effect enough and then toggle the LEDs off and on I start getting that garbage output, followed by the controller dropping the MQTT connection.

I'm not sure if the effect name string displaying twice is an issue. When I watch the state topic the MCU (when it's cooperating) reports the selected effect properly.

eldoner, 2017-04-15:
It's not even about the garbage, but it still sets LEDs to `r: 0, g: 0, b: 0`, even if it is already turned off.

bruhautomation, 2017-04-17:
Interesting. I'll check into this. So far everything I have setup has been really stable.. or at least I thought so. :P I'm checking into this and will get back to you soon.

zuidwijk, 2017-04-21:
The loop is because of this code:
`void showleds() {

  if (stateOn) {
    FastLED.setBrightness(brightness);  //EXECUTE EFFECT COLOR
    FastLED.show();
    if (transitionTime > 0 && transitionTime < 130) {  //Sets animation speed based on receieved value
      FastLED.delay(1000 / transitionTime);
      //delay(10*transitionTime);
    }
  }
  else {
    setColor(0, 0, 0);
  }
}`
if stateOn == false then setColor....

zuidwijk, 2017-04-21:
The double effect comes from void processJson() and void callback().

In processJson:
```
    if (root.containsKey("effect")) {
      effect = root["effect"];
      effectString = effect;
      Serial.print(effectString);
      twinklecounter = 0; //manage twinklecounter
    }
```

In void callback:
```
  ...
  if (!processJson(message)) { **<-- calls function processJson, causes the first effect print**
    return;
  }

  if (stateOn) {

    realRed = map(red, 0, 255, 0, brightness);
    realGreen = map(green, 0, 255, 0, brightness);
    realBlue = map(blue, 0, 255, 0, brightness);
  }
  else {

    realRed = 0;
    realGreen = 0;
    realBlue = 0;
  }
  Serial.print("callback: "); **<-- my debug**
  Serial.println(effect); **<-- second print**
  ...
```

zuidwijk, 2017-04-21:
I've debugged the code and found out why there is a loop when state is OFF and effect is !solid.

The code first checks what to do (the effect). If it is an effect (anything but solid) then it calls the function showleds(). In the showleds() it checks on status. If status is OFF then setColor(0,0,0)... continuous...

When the effect is solid, that part of the code doesn't call function showleds():

```
  if (startFade && effectString == "solid") {
    // If we don't want to fade, skip it.
    if (transitionTime == 0) {
      setColor(realRed, realGreen, realBlue);

      redVal = realRed;
      grnVal = realGreen;
      bluVal = realBlue;

      startFade = false;
    }
    else {
      loopCount = 0;
      stepR = calculateStep(redVal, realRed);
      stepG = calculateStep(grnVal, realGreen);
      stepB = calculateStep(bluVal, realBlue);

      inFade = true;
    }
  }

  if (inFade) {
    startFade = false;
    unsigned long now = millis();
    if (now - lastLoop > transitionTime) {
      if (loopCount <= 1020) {
        lastLoop = now;

        redVal = calculateVal(stepR, redVal, loopCount);
        grnVal = calculateVal(stepG, grnVal, loopCount);
        bluVal = calculateVal(stepB, bluVal, loopCount);

        if (effectString == "solid") {
          setColor(redVal, grnVal, bluVal); // Write current values to LED pins
        }
        Serial.print("Loop count: ");
        Serial.println(loopCount);
        loopCount++;
      }
      else {
        inFade = false;
      }
    }
  }
```

Maybe, instead of checking the state in showleds(), there can be an effect called "off"? Then when state is OFF, don't do any effect, go directly to "effect off"...?

zuidwijk, 2017-04-21:
Final thoughts ;-) Because the whole code is a continuous loop, is doesn't matter how the code is. When effect is set to solid, it does a setColor() without any Serial.print();. When the effect is !solid, than the showleds() function is called and there is also the setColor() function, only with a Serial.print()... remove this Serial.print(), and it doesn't loop anymore.

In my opinion the best way is to print text is in the payload function, not the continuous loop (only for debugging).

bruhautomation, 2017-04-23:
Thanks much for the suggested change, @zuidwijk. My apologies for the issue. Yeah, that bit at the end of the code is a lazy way of making the effect stick after turning off the strip in HA. When you "turn on" the strip in HA, it remembers the last effect it was running and resumes it. 

The latest version I just uploaded should resolve this issue. Let me know if it doesn't! Cheers!

closed 2017-04-23 by bruhautomation