bruhautomation/ESP-MQTT-JSON-Digital-LEDs #16: No speed controll

viperguy07, 2017-04-20:
I has trying to change the speed of an animation and nothing happened, I thought it was my config but it looks like it isn't in the arduino code

OlavAlexanderMjelde, 2017-04-23:
It works here, I'm thinking maybe you changed something in the Yaml and forgot some changes?

olavxxx, 2017-04-23:
It works here, I'm thinking maybe you changed topic and forgot some changes in the YAML file?

bruhautomation, 2017-04-23:
Sorry about the issue! So you have the slider in Home Assistant? It's just not causing an effect on the LED strip? 

Try running the command `mosquitto_sub -d -u yourMQTTusername -P yourMQTTpassword -t "#"` on the same device that hosts your MQTT broker. Do you see published info with the transition information when you move the slider in HA?

bruhautomation, 2017-05-23:
Closing issue because of inactivity. Feel free to open another one if you have issues. Cheers!

closed 2017-05-23 by bruhautomation