bruhautomation/ESP-MQTT-JSON-Digital-LEDs #17: ESP "crashes" after setting effect

zuidwijk, 2017-04-20:
I've tried the code as-is and only changed the wifi and mqtt settings.

Without connecting the ledstrip to the ESP, only to test, the network connection is lost.
I start with 4 terminals:
1 to view serial output
2 continuous ping to esp
3 send mqtt command
4 view mqtt commands returning

When setting an effect with mqtt, after a few seconds the ping reply stops. The serial output says "Attempting MQTT connection...failed, rc=-2 try again in 5 seconds"... 

When printing out the ip address in the process to reconnect mqtt, it says it still has it's ip address, whilst it isn't pingable.

I have this on my nodemcu developer board, and also on a esp-01.

bruhautomation, 2017-04-23:
Hmm, that's interesting. I am not able to reproduce this on my dev setup. Can you post the full output from the Serial monitor from when it starts up to when you're getting the error messages?

zuidwijk, 2017-04-24:
Yes, I'll make a screen recording to show what happens :)

zuidwijk, 2017-04-24:
Not able to make a recording yet, but here my first thoughts: because effects are all using the same loop (even effect "solid" is in the same loop), I thought maybe it crashes because of the many checks and print(). I stript all the effects but one: candy cane. It had no effect. It crashes once the effect starts. Sometimes within a few seconds, other times after >5 minutes.

zuidwijk, 2017-04-24:
@bruhautomation I just pended you an email within 3 links to screen recordings.

bruhautomation, 2017-06-02:
Hello. Sorry for the forever delay on this. I've been testing a couple of things and MAY be onto something. So far I'm at 24 hours and no crashes. Can you remove these lines from your code (23 and 24) and tell me if that helps the stability at all? 

```
#define FASTLED_INTERRUPT_RETRY_COUNT 0 //makes animations smoother... supposedly... 
#define FASTLED_ALLOW_INTERRUPTS 0
```

zuidwijk, 2017-06-02:
Nice, I'll try this within a few hours :)

zuidwijk, 2017-06-02:
I've used the code 1-on-1 (only wifi, mqtt and name changed). I'll let it run for at least ~7 hours to test the stability. Without the ledstrip, only the esp-01. I'll let you all know if it crashes, or keeps on running.

bruhautomation, 2017-06-02:
Sounds good! I'm at >48 hours on both my setups now. Everything seems to be running great. Hope it works for everyone!

zuidwijk, 2017-06-02:
121xx seconds (ping replies) further, about 3,5 hours further, it's still working without any errors/crashes... Looks promising so far 👍

zuidwijk, 2017-06-02:
I did notice one error... I'll look it up later in the code...

```
marcel@HomeAssistant:~$ mosquitto_pub -t 'brew/ledstrip/set' -u x -P x -m '{"color": {"r": 255, "g": 255, "b": 0}, "transition": 5}'
marcel@HomeAssistant:~$ mosquitto_pub -t 'brew/ledstrip/set' -u x -P x -m '{"color": {"r": 255, "g": 31, "b": 0}, "transition": 5}'
```

First I did color [255, 255, 0] than to color [255, 31, 0] and it goes to color [255, 0, 0].
Doing the last command again, it goes to [255, 32, 0] (the 32 instead of 31 is just a minor issue)

zuidwijk, 2017-06-02:
273xx seconds, more than 7,5 hours stable. I'm liking it... couldn't test it further... This is soo much better!

sisensee, 2017-06-02:
2.29 hours and climbing ( this is normal crash threshold ) AWESOME! Thanks for the work!

bruhautomation, 2017-06-03:
Nice! Pumped it is working for you guys. Let me know if you have any other stability issues! I'm going to close the issue once you guys give me the green light. Cheers!

zuidwijk, 2017-06-03:
One issue noticed with an esp-01. It's up and stable for about 9hrs now. Yet the effects aren't working right. For example: candy cane is showing the colors, but is flickering and not moving. Rainbow is moving, yet also flickering. I'll try to make a movie of it.

zuidwijk, 2017-06-03:
Ok, I think I've fixed it 👍 

You have to use '#define FASTLED_INTERRUPT_RETRY_COUNT 0'. Without it, it looks awful. I've added it (without '#define FASTLED_ALLOW_INTERRUPTS 0') and the effects are smooth again, no glitches, candy cane is moving again etc.

Your code with the two adjustments, can you and/or anyone else try this too?
```
/*
  .______   .______    __    __   __    __          ___      __    __  .___________.  ______   .___  ___.      ___   .___________. __    ______   .__   __.
  |   _  \  |   _  \  |  |  |  | |  |  |  |        /   \    |  |  |  | |           | /  __  \  |   \/   |     /   \  |           ||  |  /  __  \  |  \ |  |
  |  |_)  | |  |_)  | |  |  |  | |  |__|  |       /  ^  \   |  |  |  | `---|  |----`|  |  |  | |  \  /  |    /  ^  \ `---|  |----`|  | |  |  |  | |   \|  |
  |   _  <  |      /  |  |  |  | |   __   |      /  /_\  \  |  |  |  |     |  |     |  |  |  | |  |\/|  |   /  /_\  \    |  |     |  | |  |  |  | |  . `  |
  |  |_)  | |  |\  \-.|  `--'  | |  |  |  |     /  _____  \ |  `--'  |     |  |     |  `--'  | |  |  |  |  /  _____  \   |  |     |  | |  `--'  | |  |\   |
  |______/  | _| `.__| \______/  |__|  |__|    /__/     \__\ \______/      |__|      \______/  |__|  |__| /__/     \__\  |__|     |__|  \______/  |__| \__|

  Thanks much to @corbanmailloux for providing a great framework for implementing flash/fade with HomeAssistant https://github.com/corbanmailloux/esp-mqtt-rgb-led
  
  To use this code you will need the following dependancies: 
  
  - Support for the ESP8266 boards. 
        - You can add it to the board manager by going to File -> Preference and pasting http://arduino.esp8266.com/stable/package_esp8266com_index.json into the Additional Board Managers URL field.
        - Next, download the ESP8266 dependancies by going to Tools -> Board -> Board Manager and searching for ESP8266 and installing it.
  
  - You will also need to download the follow libraries by going to Sketch -> Include Libraries -> Manage Libraries
      - FastLED 
      - PubSubClient
      - ArduinoJSON
*/

//Note from @zuidwijk: This is needed for smoothnes on an ESP-01
#define FASTLED_INTERRUPT_RETRY_COUNT 0 //makes animations smoother... supposedly... 

#include <ArduinoJson.h>
#include <ESP8266WiFi.h>
#include <PubSubClient.h>
#include "FastLED.h"
#include <ESP8266mDNS.h>
#include <WiFiUdp.h>
#include <ArduinoOTA.h>

/************ WIFI and MQTT Information (CHANGE THESE FOR YOUR SETUP) ******************/
const char* ssid = "YourSSID"; //type your WIFI information inside the quotes
const char* password = "YourWIFIpassword";
const char* mqtt_server = "your.MQTT.server.ip";
const char* mqtt_username = "yourMQTTusername";
const char* mqtt_password = "yourMQTTpassword";
const int mqtt_port = 1883;

/**************************** FOR OTA **************************************************/
#define SENSORNAME "porch" //change this to whatever you want to call your device
#define OTApassword "yourOTApassword" //the password you will need to enter to upload remotely via the ArduinoIDE
int OTAport = 8266;

/************* MQTT TOPICS (change these topics as you wish)  **************************/
const char* light_state_topic = "bruh/porch";
const char* light_set_topic = "bruh/porch/set";

const char* on_cmd = "ON";
const char* off_cmd = "OFF";
const char* effect = "solid";
String effectString = "solid";
String oldeffectString = "solid";

/****************************************FOR JSON***************************************/
const int BUFFER_SIZE = JSON_OBJECT_SIZE(10);
#define MQTT_MAX_PACKET_SIZE 512

/*********************************** FastLED Defintions ********************************/
#define NUM_LEDS    186
#define DATA_PIN    5
//#define CLOCK_PIN 5
[... 898 lines cut ...]
{
  if ( random8() < chanceOfGlitter) {
    leds[ random16(NUM_LEDS) ] += CRGB(red, green, blue);
  }
}

/********************************** START SHOW LEDS ***********************************************/
void showleds() {

  delay(1);

  if (stateOn) {
    FastLED.setBrightness(brightness);  //EXECUTE EFFECT COLOR
    FastLED.show();
    if (transitionTime > 0 && transitionTime < 130) {  //Sets animation speed based on receieved value
      FastLED.delay(1000 / transitionTime);
      //delay(10*transitionTime);
    }
  }
  else if (startFade) {
    setColor(0, 0, 0);
    startFade = false;
  }
}
```

zuidwijk, 2017-06-03:
Stable for >40min :+1: 
Another bug: try an effect and than a non existing effect. The effect "freezes". Selecting an existing effect again and it works again.

bruhautomation, 2017-06-03:
Nice! Pumped you got that working. Yeah, I'll keep that line in mind if people are reporting issues with unsmooth animations. Both my 5V and 12V setups are stable >48 hours without it. I think I'll leave the code as is until I get a better grasp on why the two lines I removed were causing the crashing. I'm pretty certain the culprit was `#define FASTLED_ALLOW_INTERRUPTS 0`, but who knows. 

If you don't mind, can you open a new issue on the effect freezing? I'm sure there are a few more bugs like that I need to work out. Cheers!

closed 2017-06-03 by bruhautomation