bruhautomation/ESP-MQTT-JSON-Digital-LEDs #18: MQTT loses connection

racingpassion, 2017-04-22:
Need help, Great code but it crashes after a period of time. It loses connection with MQTT and the only way to bring it back is to reset the sensor completely only for it to last for a short period of time and do exactly the same thing.

bruhautomation, 2017-04-23:
I've noticed a similar issue on my larger LED setup.  Interestingly, when my strip crashes, it fully restarts and is responsive to new MQTT messages. 

The full sequence is, I call an effect, it works as expected for an hour or two, then randomly the strip freezes, and the strip restarts. When it restarts, it calls color (0,0,0) and the strip is off and ready to receive new messages. 

Best I can figure the issue I'm running into is a random wdt reset, not an MQTT disconnection. I added a line in my latest update to attempt to reconnect wifi and mqtt if it becomes disconnected. Give that a shot and see if that reconnects it after it disconnects.  

I'm struggling to reproduce this error on my dev setup, but I will let you know if I find a solution.

zuidwijk, 2017-04-24:
> The full sequence is, I call an effect, it works as expected for an hour or two, then randomly the strip freezes, and the strip restarts. When it restarts, it calls color (0,0,0) and the strip is off and ready to receive new messages.

This is exactly what happens with my setup (even without any leds connected to the esp).
https://github.com/bruhautomation/ESP-MQTT-JSON-Digital-LEDs/issues/17

racingpassion, 2017-04-25:
Thanks Ben for the response trying the new code seems a lot more stable staying connected and all animations seem to work fine.:)

bruhautomation, 2017-04-27:
@zuidwijk, yeah, I've noticed the exact same thing on one of my setups. It's really bothering me at this point. 👎 I have two setups of this sketch with identical code (sans the topic and name). 

The one setup, 12V leds via logic shifter, crashes every two hours or so for no apparent reason like you just described.  

> The full sequence is, I call an effect, it works as expected for an hour or two, then randomly the strip freezes, and the strip restarts. When it restarts, it calls color (0,0,0) and the strip is off and ready to receive new messages.

The second setup, 2x 5V neopixel rings is super solid and I can't get it to crash even after 24+ hours of running. 

That's a bit scary to me as it's indicative of a hardware problem or a limitation of the ESP board. I noticed my 12V strip had a loose connector and will randomly dropped out half of the strip causing the board to crash if the wind blew wrong. I've tried to fix that connector 3 times and hopefully now it's solid. I'm continuing to try and elucidate the stability issue, but I can't find a rhyme or reason yet. 

I'll keep you posted as I learn more. I want to try a long segment of 5V RGBW strips and see if they show the same problem.

sisensee, 2017-04-27:
running a 5v setup (ws2812b) with 192 LEDs. I've ran one time on a rainbow effect crashed after 2 hours and a second time on just white ran no problem. Also kind of interesting is HA registered when it shut off (crashed). Both times I didn't change any code just let my timer automation run, which is on 45 minutes before sun set and off 45 minutes after sun rise. I'm trying the rainbow effect again tonight. When it crashed last time it came back from the crash at 0,0,0 as well.

sisensee, 2017-05-02:
Same issue here, with newest code (4/27/17) 
`
Attempting MQTT connection...connected
Setting LEDs:
r: 0, g: 0, b: 0
Message arrived [ocr/sign/set] {"state": "ON"}
solid
Setting LEDs:
r: 255, g: 255, b: 255
Message arrived [ocr/sign/set] {"state": "ON", "effect": "rainbow"}
rainbow

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v3de0c112
~ld
ü
`

zuidwijk, 2017-05-03:
I've noticed in earlier experiences that an ESP crashes if you printout too much info.
For example, I used to use LUA code, and with to many counters (3x 0-255 for color fading).
While printing out each status, the ESP crashes. Yet with no printing, it worked fine.

If tried to add a `#define DEBUG = false;` and for each Serial.println or Serial.print I added `if (DEBUG)`

It did its job: no printing, yet the ESP keeps on crashing :(

I see you've updated the code. I'm gonna try that again.

sisensee, 2017-05-03:
From what I have read (sorry not a code guy by nature) is that this isnt limited to this! Every possibility from a memory heap error in the library. 
https://github.com/esp8266/Arduino/blob/master/doc/faq/a02-my-esp-crashes.md
and 
here: https://github.com/esp8266/Arduino/issues/1017 
Sadly I haven't come across a solution yet. I did read some interesting thoughts on adding a delay in the loop and how that frees up the memory heap error. Mentioned in the second link I believe.

hagensieker, 2017-05-23:
I just added a new issue about SPI Pins and Fast LED.  I flashed a new NodeMCU to rule out hardware problems and connected to serial monitor and ran "Police All at max speed and brightness.  Crashes in 5 minutes or less with lights hooked up or not.  Serial monitor says MQTT disconnected and retries but cannot reconnect.  Resetting sets lights to 0 and makes it ready.

Ran the FastLED Demo 100 with no issues noted.

bruhautomation, 2017-05-23:
Yeah, I'm continuing to explore this problem. Sadly, I've noticed a chronic instability with the NodeMCU chips. A lot of people are reporting random shutdowns. I think most of the time the chip restarts okay. For things like the sensor node, an occasional reset has no consequences, but for these LED strips it causes the strip to shut off which is obviously noticeable.  

I can say that my setup with 5V LED neopixel rings is 100% rock solid. I can't get it to crash. It's not clear to me why the combination of a logic level shifter and the longer LED strip seems to cause the instability. 

I recently just got an ESP32 NodeMCU proto board. I'm going to give that a shot and see if the extra horsepower helps. I'll also continue to look at the code and see if there is anything I can do to optimize. 

Cheers!

sisensee, 2017-05-23:
You could try to switch the NodeMCU being used in each example and see if its a possible hardware issue. Take your 5v node and switch it with the 12v one.

hagensieker, 2017-05-23:
Thanks for the quick response.  I was going to try another chip like Feather or Feather Huzzah to see if that had any effect at all later.  Have you gone down that route?  Or is it ESP8266 in general related?

zuidwijk, 2017-05-23:
I've tried it on three esp-boards: 2 different esp-01 and one [nodemcu development board](http://s4.electrodragon.com/wp-content/uploads/2015/04/NodeMCU-LUA-Amica-R2-04.jpg) (v1.0).
They all crashes. I'm gonna try it on a Wemos D1 mini.

zuidwijk, 2017-05-23:
Wemos D1 crashed to... 

```
marcel@ha:~$ mosquitto_pub -t 'bruh/porch/set' -m '{"state": "ON"}'
marcel@ha:~$ mosquitto_pub -t 'bruh/porch/set' -m '{"effect": "solid"}'
marcel@ha:~$ mosquitto_pub -t 'bruh/porch/set' -m '{"effect": "candy cane"}'
```
I started a ping once the last commando was given:
```
64 bytes from 10.0.0.167: icmp_seq=270 ttl=128 time=4.960 ms
64 bytes from 10.0.0.167: icmp_seq=271 ttl=128 time=1.475 ms
64 bytes from 10.0.0.167: icmp_seq=272 ttl=128 time=3.377 ms
Request timeout for icmp_seq 273
Request timeout for icmp_seq 274
Request timeout for icmp_seq 275
```
So within 5 minutes (~272 seconds), it crashed:
```
WiFi connected
IP address:
10.0.0.167
Ready
IP Address: 10.0.0.167
Attempting MQTT connection...connected
Setting LEDs:
r: 0, g: 0, b: 0
Message arrived [bruh/porch/set] {"state": "ON"}
solid
Setting LEDs:
r: 255, g: 255, b: 255
Message arrived [bruh/porch/set] {"effect": "solid"}
solid
Setting LEDs:
r: 255, g: 255, b: 255
Message arrived [bruh/porch/set] {"effect": "candy cane"}
candy cane
Attempting MQTT connection...failed, rc=-2 try again in 5 seconds
Attempting MQTT connection...failed, rc=-2 try again in 5 seconds
Attempting MQTT connection...failed, rc=-2 try again in 5 seconds
Attempting MQTT connection...failed, rc=-2 try again in 5 seconds
Attempting MQTT connection...failed, rc=-2 try again in 5 seconds
Attempting MQTT connection...failed, rc=-2 try again in 5 seconds
Attempting MQTT connection...failed, rc=-2 try again in 5 seconds
Attempting MQTT connection...failed, rc=-2 try again in 5 seconds

Exception (29):
epc1=0x4000e1b2 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: sys
sp: 3ffffd80 end: 3fffffb0 offset: 01a0

>>>stack>>>
3fffff20:  00000002 0000002c 3fff0bbc 401004f4
3fffff30:  402179c5 00000002 3fff0bbc 40107254
3fffff40:  40237d70 3fff0bbc 60000600 402179e3
3fffff50:  40218ddd 3fff0bc0 3ffee510 23c7ed93
3fffff60:  60000600 00000000 3ffed6cc 00000000
3fffff70:  402214e9 3fff0bc0 3ffee510 23c7d940
3fffff80:  4022152e 3fffdab0 00000000 3fffdcb0
3fffff90:  3ffee538 3fffdab0 00000000 40224862
3fffffa0:  40000f49 40000f49 3fffdab0 40000f49
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v3de0c112
~ld
   �
Connecting to SSID
......
WiFi connected
IP address:
10.0.0.167
Ready
IP Address: 10.0.0.167
Attempting MQTT connection...connected
Setting LEDs:
r: 0, g: 0, b: 0
```

zuidwijk, 2017-05-23:
If I only use solid with an color, it crashes to :(
```
marcel@ha:~$ mosquitto_pub -t 'bruh/porch/set' -m '{"state": "ON", "transition": 1, "color": {"r":255,"g":255,"b":255}'
marcel@ha:~$ mosquitto_pub -t 'bruh/porch/set' -m '{"state": "ON", "transition": 1, "color": {"r":255,"g":255,"b":255}}'
marcel@ha:~$ mosquitto_pub -t 'bruh/porch/set' -m '{"state": "ON", "transition": 1, "color": {"r":255,"g":31,"b":0}}'
marcel@ha:~$ mosquitto_pub -t 'bruh/porch/set' -m '{"state": "ON", "transition": 1, "color": {"r": 255,"g": 31,"b": 0}}'
marcel@ha:~$ mosquitto_pub -t 'bruh/porch/set' -m '{"state": "ON", "transition": 1, "color": {"r": 255,"g": 31,"b": 0}}'
marcel@ha:~$ mosquitto_pub -t 'bruh/porch/set' -m '{"state": "OFF", "transition": 1, "color": {"r": 255,"g": 31,"b": 0}}'
marcel@ha:~$ mosquitto_pub -t 'bruh/porch/set' -m '{"state": "ON", "transition": 1, "color": {"r": 255,"g": 31,"b": 0}}'
```
```
64 bytes from 10.0.0.167: icmp_seq=238 ttl=128 time=9.760 ms
64 bytes from 10.0.0.167: icmp_seq=239 ttl=128 time=16.008 ms
64 bytes from 10.0.0.167: icmp_seq=240 ttl=128 time=7.861 ms
Request timeout for icmp_seq 241
Request timeout for icmp_seq 242
Request timeout for icmp_seq 243
```
```
r: 254, g: 32, b: 0
Setting LEDs:
r: 255, g: 32, b: 0
Setting LEDs:
r: 255, g: 32, b: 0
Setting LEDs:
r: 255, g: 32, b: 0
Setting LEDs:
r: 255, g: 32, b: 0
Setting LEDs:
r: 255, g: 32, b: 0
Attempting MQTT connection...failed, rc=-2 try again in 5 seconds
Attempting MQTT connection...failed, rc=-2 try again in 5 seconds
Attempting MQTT connection...failed, rc=-2 try again in 5 seconds
Attempting MQTT connection...failed, rc=-2 try again in 5 seconds
Attempting MQTT connection...failed, rc=-2 try again in 5 seconds
Attempting MQTT connection...failed, rc=-2 try again in 5 seconds
Attempting MQTT connection...failed, rc=-2 try again in 5 seconds

Exception (29):
epc1=0x4021510b epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000019 depc=0x00000000

ctx: sys
sp: 3ffffca0 end: 3fffffb0 offset: 01a0

>>>stack>>>
3ffffe40:  3ffe931c 401015df 04000000 40101cb2
3ffffe50:  3ffffe70 4021521b 3ffed6cc 00000000
3ffffe60:  00000000 00000000 00000000 40102d0a
3ffffe70:  00000018 3ffeddec 3ffe90a6 3ffe90a6
3ffffe80:  3fff136f 3fff13f4 3fff0bbc 00000007
3ffffe90:  40216dbb 3ffedda0 3ffede78 00008000
3ffffea0:  00000001 3fff0bbc 3ffebb40 00000000
3ffffeb0:  40216a70 40216a44 00000011 ffffffff
3ffffec0:  40218825 00000000 00000020 00000000
3ffffed0:  3ffed7a4 402187ec 00000000 00000012
3ffffee0:  40216a3a 402168df 00000002 3ffedda0
3ffffef0:  402169e1 00000000 00000012 00008000
3fffff00:  40216ed4 00000000 3ffedda0 00008000
3fffff10:  4021695c 3ffed840 3fff0bbc 40237d70
3fffff20:  3ffed7a4 00000000 40218e54 40218c0c
3fffff30:  00008000 3ffee510 23552fb1 60000600
3fffff40:  40218ae6 3fff0bc0 00100000 3fffdab0
3fffff50:  40218dc6 3fff0bc0 3ffee510 23552fb1
3fffff60:  3fff0bbc 40218de6 3ffed6cc 00000000
3fffff70:  402214e9 3fff0bc0 3ffee510 23552814
3fffff80:  4022152e 3fffdab0 00000000 3fffdcb0
3fffff90:  3ffee538 00000000 3fffb724 40224862
3fffffa0:  40000f49 40000f49 3fffdab0 40000f49
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v3de0c112
~ld
   �
Connecting to SSID
..
WiFi connected
IP address:
10.0.0.167
Ready
IP Address: 10.0.0.167
Attempting MQTT connection...connected
Setting LEDs:
r: 0, g: 0, b: 0
```

hagensieker, 2017-05-24:
I kind of sort of did my home install this morning and so far (6-7 hours in) solid colors are fine and no random reboots or problems. 

I do have some other issues.  When testing on the bench the LED strips behaves perfectly where the signal pin is 3" long or so (directly off the 12 to 5vdc converter).  Get below say 12.3 volts and the first three LED's light up.  12.3 and a little up...perfect.

I drilled through an 8 inch wall, came down the garage ceiling with the 3 wire cable and mounted the power supply, Nodemcu, and 12 to 5 PS.  Maybe 2-3' of cable.  No matter the voltage the first three LED's light up.  My "extension" wires are thinner ga. copper so maybe that's got something to do with it.   Nothing Opaque tape can't fix though :)

Also when the data cable was longer the color calibration in home assistant was WAY off.  But I could still click around and find colors, not a big-ee. 

I mounted the PS near the ceiling to short the cables and my means of cutting a foot or so off the signal wire I got my color correction back.  Colors now match the click swatch in Home Assistant. 

I'm thinking Logic Shifter here might fix this but good. 

I think maybe I need to make up a new cable with thicker wire and keep it short. 

And again NO CRASHES, Random Reboots, MQTT or WiFi interruptions if I use solid colors.  I am not an animation kind of guy anyway I just wanted the LED strip for decorative lighting.   My expectations were only solid colors anyway.   Would be cool to show off though.

bruhautomation, 2017-06-02:
Okay, so I've been doing some testing on this. I think I maybe found a potential way to reduce the random resets. I've been testing my strip for ~24 hours and I've had no crashes so far. 

Can you guys remove lines 23 and 24 from your code and tell me if that helps with the random crashes? Should be these... 
```
#define FASTLED_INTERRUPT_RETRY_COUNT 0 //makes animations smoother... supposedly... 
#define FASTLED_ALLOW_INTERRUPTS 0
```

hagensieker, 2017-06-02:
I don't have a lot of longevity testing here but I'm up to about 45 minutes and it hasn't crashed yet.  Never got more than 5 mins or so before, and way less when I was changing patterns and animation speeds.  Thanks!  

Link to my setup with trials and travails.   Ben there is mention of you in my page if you want me to remove or change anything just let me know.

[http://www.hagensieker.com/blog/page/?post_id=58&title=led-strip-lighting](url)

bruhautomation, 2017-06-03:
Sweet! Nice project! look great. Let me know if you come across any more bugs. I'll do my best to help as I can. Cheers!

closed 2017-06-03 by bruhautomation