bruhautomation/ESP-MQTT-JSON-Digital-LEDs #19: Color green and blue mixed up?

olavxxx, 2017-04-23:
Sending "color":"green" gives a blue light and this is printed in the console:
`{"entity_id":"light.pixelring_strip",
"brightness":150,
"color_name":"green",
"transition":"0"
}`
`
Message arrived [sensor/neopixelring/set] {"transition": 0, **"color": {"r": 0, "b": 0, "g": 128},** "state": "ON", "brightness": 150}
solid
Setting LEDs:
r: 0, g: 75, b: 0
`

If I send "color":"blue", it gives green lends and this is printed in the console:
`{"entity_id":"light.pixelring_strip",
"brightness":150,
"color_name":"blue",
"transition":"2"
}`
`
Message arrived [sensor/neopixelring/set] {"transition": 0, "color": **{"r": 0, "b": 255, "g": 0}**, "state": "ON", "brightness": 150}
solid
Setting LEDs:
r: 0, g: 0, b: 150
`

I have to go to bed now, so I dont have any more time to look at the issue at the moment.
However I think it seems as the recieved json object has G and B mixed up, eg:
Blue becomes: R G B
Green becomes: R B G 

"color":"red" works fine, also "rgb-color":[0,255,0] etc. work fine. 

Here is the red one that works:
`Message arrived [sensor/neopixelring/set] {"transition": 0, **"color": {"r": 255, "b": 0, "g": 0},** "state": "ON", "brightness": 150}
solid
Setting LEDs:
r: 150, g: 0, b: 0
`

bruhautomation, 2017-04-23:
Check out line 65  `#define COLOR_ORDER BRG` your LED strip might have a different color layout. Sounds like it might need to be changed to `#define COLOR_ORDER GRB` Cheers!

olavxxx, 2017-04-24:
Yes, that was correct 👍 
Maybe the config line should have a comment on how you check what setting is correct?

closed 2017-04-24 by olavxxx