bruhautomation/ESP-MQTT-JSON-Digital-LEDs #21: Is some of this code missing?

pushscooter, 2017-05-09:
Slowly but surely making progress on this awesome project. I am totally new to all this and it has been quite overwhelming. Still, I am pushing on and stoked. I have uploaded everything to my node, but can't see if works because I am missing some choices on my HA states page. 

QUESTION: Is it me or is the current version of the Example Home Assistant Configuration.yaml missing a lot of lines? My HA screen is missing an effects list,  on/off switch, and color selection. Can anyone check/ update what is missing please?

this is what I have so far from mixing and matching between the lines Ben has on the original video and the YAML.file on here. Any help/advice/suggestions would be hugely appreciated:

(MQTT stuff. Then:)

light:
  - platform: mqtt_json
    name: "Porch Strip"
    state_topic: "bruh/porch"
    command_topic: "bruh/porch/set"
    rgb_state_topic: "bruh/mqttstrip/setcolorpub"
    rgb_comand_topic: "bruh/mqttstrip/setcolor"
    brightness_state_topic: "bruh/mqttstrip/setbrightnesspub"
    brightness_command_topic: "bruh/mqttstrip/setbrightnes"
    effect: true

input_select:
  porch_led_effect:
    name: "Porch LED Effect"
    options:
      - bpm
      - candy cane  
      - confetti  
      - cyclon rainbow  
      - dots  
      - fire  
      - glitter  
      - juggle  
      - lightning
      - noise  
      - police all  
      - police one  
      - rainbow  
      - rainbow with glitter  
      - ripple  
      - sinelon  
      - solid  
      - twinkle  
    brightness: true
    flash: true
    rgb: true
    optimistic: false
    qos: 0

input_slider:
  porch_animation_speed:
    name: Porch Animation Speed
    initial: 150
    min: 1
    max: 150
    step: 10

automation: 
  - alias: "Porch Input Effect"
    initial_state: True
    hide_entity: False
    trigger:
      - platform: state
        entity_id: input_select.porch_led_effect
    action:
      - service: mqtt.publish
        data_template:
          topic: "bruh/porch/seteffect"
          payload: '{"transition":{{ trigger.to_state.state | string }}}' 
          
  - alias: "Porch Animation Speed"
    initial_state: True
    hide_entity: False
    trigger:
      - platform: state
        entity_id: input_slider.porch_animation_speed
    action:
      - service: mqtt.publish
        data_template:
          topic: "bruh/porch/set"
          payload: '{"transition":{{ trigger.to_state.state | int }}}'

pushscooter, 2017-05-10:
I have been playing with this a ton and made a little progress. Here is the current view and code I have on HA:
What am I missing? Thanks in advance.

[image]

    
light:
  - platform: mqtt_json
    name: "Porch Strip"
    state_topic: "bruh/porch"
    command_topic: "bruh/porch/set"
    brightness: true
    flash: true
    rgb: true
    optimistic: false
    qos: 0
    effect: true
 
input_slider:
  porch_animation_speed:
    name: Porch Animation Speed
    initial: 150
    min: 1
    max: 150
    step: 10

input_select:
  light_mode:
    name: Porch Input Effect
    options:
     - bpm
     - candy cane  
     - confetti  
     - cyclon rainbow  
     - dots  
     - fire  
     - glitter  
     - juggle  
     - lightning
     - noise  
     - police all  
     - police one  
     - rainbow  
     - rainbow with glitter  
     - ripple  
     - sinelon  
     - solid  
     - twinkle    
    
automation: 
  - alias: "Porch Animation Speed"
    initial_state: True
    hide_entity: False
    trigger:
      - platform: state
        entity_id: input_slider.porch_animation_speed
    action:
      - service: mqtt.publish
        data_template:
          topic: "bruh/porch/set"
          payload: '{"transition":{{ trigger.to_state.state | int }}}'

bruhautomation, 2017-05-11:
Hi @pushscooter, 

Nope, everything you need to run this code is in the sample HA configuration in this repo. There was a switch from the MQTT Light component to MQTT JSON Light which is probably what is confusing you. The configuration for the MQTT JSON light is much simpler and involves sending all the commands together in the same string which the NodeMCU can parse and utilize. 

The effect dropdown will show up in the card for the light after you turn on the toggle.

Cheers,
Ben

pushscooter, 2017-05-11:
Gotcha, thanks so much for explaining that Ben! 
Just one question though. How do I "turn on the toggle"? And what do you mean by "CARD"? (ok,ok, thats two questions...)

I am so close I can feel it! Thanks again for all your help Ben! 
Dan
PS: This is what my HA looks like after making the changes you suggested.
[image]

pushscooter, 2017-05-12:
I have found the resources that show examples, but I still feel lost (even more so actually). Can you (or anyone) specifically show me what to add to my YAML.file to show the LED on off switch (is that what you mean by toggle?) and the effect list?

pushscooter, 2017-05-12:
Got it, kind of. Thanks soooo much for your help. I set up some groups and a card for this LED strip via your awesome videos, but the effect dropdown still is not showing up. Is there a difference between setting up a toggle and setting up a switch. If so, can you please show what I should add to my YAML to ad a toggle. I have been looking everywhere and can't find it. Here is my YAML file and what my HA looks like: 
Thanks again for everything. I know I am sooooo close!!

[image]
[image]

homeassistant:

mqtt:
  broker: m13.cloudmqtt.com
  port: 15366
  client_id: home-assistant-1
  username: 
  password: 
  protocol: 3.1     
    
light:
  - platform: mqtt_json
    name: "Porch Strip"
    state_topic: "bruh/porch"
    command_topic: "bruh/porch/set"
    effect: true
    effect_list:
      - bpm
      - candy cane  
      - confetti  
      - cyclon rainbow  
      - dots  
      - fire  
      - glitter  
      - juggle  
      - lightning
      - noise  
      - police all  
      - police one  
      - rainbow  
      - rainbow with glitter  
      - ripple  
      - sinelon  
      - solid  
      - twinkle  
    brightness: true
    flash: true
    rgb: true
    optimistic: false
    qos: 0
    
switch:
  - platform: mqtt
    name: "light"
    command_topic: "bruh/porch/switch/set"     

input_slider:
  porch_animation_speed:
    name: Porch Animation Speed
    initial: 150
    min: 1
    max: 150
    step: 10

automation: 
  - alias: "Porch Animation Speed"
    initial_state: True
    hide_entity: False
    trigger:
      - platform: state
        entity_id: input_slider.porch_animation_speed
    action:
      - service: mqtt.publish
        data_template:
          topic: "bruh/porch/set"
          payload: '{"transition":{{ trigger.to_state.state | int }}}'        
                   
group:     
  airstream:
    name: addressable leds 
    icon: mdi:truck-trailer
    view: yes
    entities:
      - group.addressable_leds
  environment:
    name: weather
    icon: mdi:weather-windy-variant
    view: yes
    entities:
  addressable leds:
    name: addressable leds
    entities:
      - switch.light
      - input_slider.porch_animation_speed

pushscooter, 2017-05-13:
Got it... kind of. Thanks soooo much @bruhautomation  for your help. I set up some groups and a card for this LED strip via your awesome videos, but the effect dropdown still is not showing up. 

 QUESTION:
     Is there a difference between setting up a toggle and setting up a switch. If so, can you please specifically show me the lines I should add to my YAML to add a toggle to access the lights. I have been looking everywhere and can't find it. Just info about switches. And my switch doesn't turn on my LEDs, or offer color selection, or the effects list. 

Here is my YAML file and what my HA pages look like:
Thanks again for everything. I know I am sooooo close and so excited to finally make this happen!!

[image]
[image]

`homeassistant:
mqtt:
  broker: m13.cloudmqtt.com
  port: 15366
  client_id: home-assistant-1
  username: 
  password: 
  protocol: 3.1     
    
light:
  - platform: mqtt_json
    name: "Adressable LEDs"
    state_topic: "airstream/addressable_leds"
    command_topic: "airstream/addressable_leds/set"
    effect: true
    effect_list:
      - bpm
      - candy cane  
      - confetti  
      - cyclon rainbow  
      - dots  
      - fire  
      - glitter  
      - juggle  
      - lightning
      - noise  
      - police all  
      - police one  
      - rainbow  
      - rainbow with glitter  
      - ripple  
      - sinelon  
      - solid  
      - twinkle  
    brightness: true
    flash: true
    rgb: true
    optimistic: false
    qos: 0
 
switch:
  - platform: mqtt
    name: "on off"
    command_topic: "airstream/addressable_leds/switch/set"     

input_slider:
  addressable_leds_animation_speed:
    name: Adressable LEDs Animation Speed
    initial: 150
    min: 1
    max: 150
    step: 10

automation: 
  - alias: "Adressable LEDs Animation Speed"
    initial_state: True
    hide_entity: False
    trigger:
      - platform: state
        entity_id: input_slider.addressable_leds_animation_speed
    action:
      - service: mqtt.publish
        data_template:
          topic: "airstream/addressable_leds/set"
          payload: '{"transition":{{ trigger.to_state.state | int }}}'        
                   
group:     
  airstream:
    name: addressable leds 
    icon: mdi:truck-trailer
    view: yes
    entities:
      - group.addressable_leds
  environment_view:
    name: weather
    icon: mdi:weather-windy-variant
    view: yes
    entities:
      - group.environment
  addressable leds:
    name: addressable leds
    entities:
      - switch.on_off
      - input_slider.addressable_leds_animation_speed   
  environment:
    name: environment
    entities:
      - sensor.yr_symbol
      - sun.sun
      `

pushscooter, 2017-05-18:
got it figured out finally. So relieved!!!
The problem was trying to do this from my desktop

bruhautomation, 2017-05-23:
Sweet! Glad you got it working. Sorry again for missing your earlier message. The pings I get from GitHub often miss my inbox so I don't see them until I check back. Cheers!

closed 2017-05-23 by bruhautomation