bruhautomation/ESP-MQTT-JSON-Digital-LEDs #22: Can't find the effect dropdown or toggle

pushscooter, 2017-05-17:
Hi @bruhautomation , What do you mean when you say, "The effect dropdown will show up in the card for the light after you turn on the toggle."?

Can you please specifically explain both where and how to turn on the toggle. 
Do I need to add any lines of code? I think I am overthinking this step which is probably super simple. Can you please help and provide as much detail as possible. I have tried every single command (is that what you call them?) in the developer tools/ services page of HA. 

Thanks

sisensee, 2017-05-17:
When the light is turned on the options for the effects show up. If the light is off then you do not see the options for the effects.

pushscooter, 2017-05-17:
Awesome @sisensee What I am trying to figure out is speciffically HOW to turn on/ toggle the light. Looking for speciffic step by step directions to a simple step that has been driving me crazy. Can you help?

pushscooter, 2017-05-17:
Right now there is no option for me to "turn on" the light. This is all I see.
[image]

sisensee, 2017-05-17:
This is your default view? Can you sanitize your groups.yamls and configuration.yaml (remove passwords and IP addresses) then post them. Mark them down by putting a three tick marks above the code and below. (the one on the tilde key, I can't find it on my iPhone keyboard). I'll see if I can help.

pushscooter, 2017-05-18:
WOW!!!! I have been trying to do this from my computer and having nothing but problems. I finally got my Raspberry 3 today in the mail and within a few hours was able to get everything working. Now simply turning on the light makes sense because it is an available option. 
Thanks so much for offering your help @sisensee even though I ended up figuring it out. I still have some minor issues to play with but it feels soooo good to get this working. 

I DO NOT RECOMMEND ATTEMPTING TO MAKE THIS WORK WITHOUT A PI AS YOUR SERVER. SO MUCH TIME SPENT IN AGONIZING FRUSTRATION TRYING TO DO THIS FROM A DESKTOP. YES, it made a huge difference switching over.

sisensee, 2017-05-18:
Glad you got it working man. Its always an awesome feeling when you get something to start working that has been plaguing you!

pushscooter, 2017-05-20:
You said it! I thought I was going crazy for a while there. Thanks again for reaching out @sisensee!

bruhautomation, 2017-05-23:
Sorry to not be more help on this issue! Pumped you got it working, @pushscooter! Yeah, using a Pi makes life a lot easier. Feel free to open another issue if you have any other problems. I'll do my best to be more responsive. Cheers!

closed 2017-05-23 by bruhautomation