bruhautomation/ESP-MQTT-JSON-Digital-LEDs #24: No Hardware SPI Pins Defined

hagensieker, 2017-05-23:
Latest code from here, latest FastLED.  When compiling the sketch for NodeMCU I get the following:

`In file included from /Users/john/Documents/Arduino/ESP_MQTT_Digital_LEDs/ESP_MQTT_Digital_LEDs.ino:29:0:
/Users/john/Documents/Arduino/libraries/FastLED/FastLED.h:17:21: note: #pragma message: FastLED version 3.001.003
 #    pragma message "FastLED version 3.001.003"
                     ^
In file included from /Users/john/Documents/Arduino/libraries/FastLED/FastLED.h:65:0,
                 from /Users/john/Documents/Arduino/ESP_MQTT_Digital_LEDs/ESP_MQTT_Digital_LEDs.ino:29:
/Users/john/Documents/Arduino/libraries/FastLED/fastspi.h:110:23: note: #pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output
 #      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"
                       ^
Sketch uses 264731 bytes (25%) of program storage space. Maximum is 1044464 bytes.
Global variables use 36100 bytes (44%) of dynamic memory, leaving 45820 bytes for local variables. Maximum is 81920 bytes.
Uploading 268880 bytes from /var/folders/tc/j87mr7r54r5_c2vr57wnwvzr0000gn/T/arduino_build_578265/ESP_MQTT_Digital_LEDs.ino.bin to flash at 0x00000000
................................................................................ [ 30% ]
................................................................................ [ 60% ]
................................................................................ [ 91% ]
.......................                                                          [ 100% ]
`

However it does compile and does install and work.  However the longest run I get is maybe 15 minutes tops.  The strip either just shuts off, or freezes mid animation.  A reset of NodeMCU clears it and restarts to Solid White.  Haven't logged via serial monitor yet but seems to be replicating other issues known issues in the github issues section.

pushscooter, 2017-05-28:
I am having this same issue, if anyone figures out how to resolve this issue please let us know.

bruhautomation, 2017-06-02:
Hello! So I've been doing some testing trying to resolve this issue. Can you guys remove these lines from your code, reupload, and let me know if that helps your stability? 

It's doing the trick on my 12v setup today. 

https://github.com/bruhautomation/ESP-MQTT-JSON-Digital-LEDs/blob/master/ESP_MQTT_Digital_LEDs/ESP_MQTT_Digital_LEDs.ino#L23-24

bruhautomation, 2017-06-02:
For the record that's supposed to be both lines 23 and 24. 

```#define FASTLED_INTERRUPT_RETRY_COUNT 0 //makes animations smoother... supposedly... 
#define FASTLED_ALLOW_INTERRUPTS 0```

pushscooter, 2017-06-02:
Woah!!!!!! That is WAY better! I have been playing with it all night and the only issue I have run into so far is that the Cyclon Rainbow effect freezes when adjusting the animation speed. 
Thanks so much for checking in on this and making the suggestion. 

Soooo stoked this is working better!

bruhautomation, 2017-06-03:
Nice! Pumped to find a fix. I'll look into the cyclon rainbow issue and see if I can find a fix as I have time. Cheers!

closed 2017-06-03 by bruhautomation