bruhautomation/ESP-MQTT-JSON-Digital-LEDs #26: Effects stop after a while

g3n35i5, 2017-05-26:
Solid light are working great but all effects stop working after a while (between 5 and 10 minutes) and the leds become black. Did anyone else encounter this problem?

Thank you for your support!

bruhautomation, 2017-06-02:
Hello! Sorry for delay getting back to you. I've been doing some testing. 

Can you remove these lines from your code and tell me if that fixes your stability? 

https://github.com/bruhautomation/ESP-MQTT-JSON-Digital-LEDs/blob/master/ESP_MQTT_Digital_LEDs/ESP_MQTT_Digital_LEDs.ino#L23-24

It's doing the trick on my 12v setup.

bruhautomation, 2017-06-02:
For the record, that's supposed to be lines 23 and 24. 
```
#define FASTLED_INTERRUPT_RETRY_COUNT 0 //makes animations smoother... supposedly... 
#define FASTLED_ALLOW_INTERRUPTS 0
```

bruhautomation, 2017-06-03:
Seems like people are reporting that the fix worked. Feel free to open another issue if you continue to have problems. Cheers!

closed 2017-06-03 by bruhautomation

andrein81, 2017-07-01:
by removing #define FASTLED_ALLOW_INTERRUPTS 0 i have some random flickering on the strip.