bruhautomation/ESP-MQTT-JSON-Digital-LEDs #27: Test run via Mqtt Tools

bidyutper, 2017-05-30:
I have successfully uploaded the code but failed to test run with mqtt tools like mqttLens, Fx or other mobile mqtt client app...what are the parameter i should pass from pub side.. tried pub on topic "light_set_topic" and 
message {"entity_id":"light.porch_strip",
"color_name":"green",
"brightness":255,
"flash":"short"
}

obviously not  correct one..no response from node mcu

bruhautomation, 2017-05-30:
The JSON commands I list in the readme are the ones you'd use with the Home Assistant light.turn_on service. Essentially, it manipulates that color_name/flash shorthand into a new MQTT message it sends to the NodeMCU. The actual transmitted MQTT message should look something like..  

```
  {
    "brightness": 120,
    "color": {
      "r": 255,
      "g": 100,
      "b": 100
    },
    "flash": 2,
    "transition": 5,
    "state": "ON"
  }
```

Cheers!

bidyutper, 2017-05-31:
k..thx

closed 2017-05-31 by bidyutper

andrein81, 2017-06-09:
I try to send this :  {
    "brightness": 120,
    "color": {
      "r": 255,
      "g": 100,
      "b": 100
    },
    "flash": 2,
    "transition": 5,
    "state": "ON"
  }

via MQTTBOX to the topic bruh/porch/set

and nothing happens.

Can I get a little help ? :)

Thank you