bruhautomation/ESP-MQTT-JSON-Digital-LEDs #28: Very weak lights. NOT WORKING AT ALL

thundergreen, 2017-05-31:
Hi guys.. NIce prject. I setup everything according to the tutorials here. I also get disconnects and my LED strip is just glewing weak red. I can see that it might get signal cause if i chose an effect lights start to flicker. But No color available nothing. 

I dunno where to start troubleshooting. I have NodeMCU 1 with ws2811 strip.

Help appreciated :)

bruhautomation, 2017-06-02:
@thundergreen, hmm, lots of stuff could be going on. The first thing I'd do is try a couple of the FastLED demo sketches. They will help determine if your hardware is working right. My code uses MQTT and has to shake hands for everything to turn on correctly. 

If the demo sketches are working, double check all your connections. That you're going in and out of the correct sides of your logic level shifter. Make sure your code is set to the right kind of LED and the LED order is correct. Some are BRG or GRB or RGB. There is no standard order and with crappy documentation, you might just have to use trial and error until you find the right order.

thundergreen, 2017-06-17:
my fault. Wrong power supply :)

closed 2017-06-17 by thundergreen