bruhautomation/ESP-MQTT-JSON-Digital-LEDs #31: Mqtt

andrein81, 2017-06-08:
If I use your script and didn't change any mqtt topics, how will a mqtt message should look to trigger one effect ?

hamidzr, 2017-07-07:
use the mosquittoclient and listen to the same topic as the one that the nodemcu is listening to. 
send some commands to ur led strip through homeasssistant.
the answer will be in the logs from mosquittoclient.

zamzon, 2017-09-25:
I am not using homeassistant but openhab with MQTT

Is there an other way to find out how to trigger the effects with MQTT commands ?

ahome-wugu, 2018-02-09:
Can someone show me what's the message json format should be sent to MQTT broker from mqtt client if using mosquitto_pub for debugging purpose ( my broker working ok, nodemcu has been properly upload this firmware ), e.g. to trigger "rainbow" effect and rest setting, pls kindly give me a message example, thank you !

Aejoh, 2018-08-05:
mosquitto_pub  -t 'SmartHome/Office/LED/set' -m '{"effect":"sine hue"}'