bruhautomation/ESP-MQTT-JSON-Digital-LEDs #32: Add some Lighteffects :)

thundergreen, 2017-06-17:
Hi Mate I finally got it running. Yeah awesome!! My wife wow effect was just amazing! Thanks mate!

after googleling a bit i found that the author of some lighteffects u used also made some oter effects

https://github.com/atuline/FastLED-Demos

Would u mind to integrate them? I didn't really understand the code.

Would be awesome !!!

THanks for this more than awesome work!!!

Greeting from a happy happy man and also a happy woman from France :) 👍

encor50, 2017-10-10:
i am working on it will post my code as soon as i finish probably this week :)

thundergreen, 2017-10-11:
Great news! Thanks! In the meanwhile. Did u already see this project:

https://github.com/doctormord/Responsive_LED_Control

KodiMkd, 2017-10-12:
ok so where is the code WITH new added effects??

encor50, 2017-10-12:
There is 2 links you have to add these effects to the brug code and the esp code just upload jt to your esp ..

KodiMkd, 2017-10-12:
PLease,please lead the blind man.
Which code goes where

encor50, 2017-10-12:
HERE IS THE CODE FOR THE ESP 
https://pastebin.com/RMPasRAU
AND ADD THESE TO YOUR HOME ASSISTANT CONFIG FILE IN ODDITION TO THE ONES FROM BRUH AUTOMATION  https://pastebin.com/ZLmpYK87

KodiMkd, 2017-10-12:
So I just remove the whole old code and add the new one from your file or I'm just adding stuff from the new code into the old one

encor50, 2017-10-12:
On your ha config file you have the some effects add the missing ones the your old code and the esp code just use arduino ide and upload it ofc fill your wifi settings in it

KodiMkd, 2017-10-12:
Not that i understood your last comment...
I just add the "new effects" in my HA config.yaml and esp file just tree place the whole code with your code. Is this corect?

encor50, 2017-10-12:
yes add the extra effect to your config.yaml and on your esp8266 (nodemcu) upload this https://pastebin.com/RMPasRAU
and dont forget to fill your wifi settings : 

const char* ssid = "YourSSID"; //type your WIFI information inside the quotes
const char* password = "YourWIFIpassword";
const char* mqtt_server = "your.MQTT.server.ip";
const char* mqtt_username = "yourMQTTusername";
const char* mqtt_password = "yourMQTTpassword";
const int mqtt_port = 1883;
 
 
 
/**************************** FOR OTA **************************************************/
#define SENSORNAME "porch" //change this to whatever you want to call your device
#define OTApassword "yourOTApassword" //the password you will need to enter to upload remotely via the ArduinoIDE
int OTAport = 8266;

KodiMkd, 2017-10-12:
i got 1 strip (planing to add 2 more) so thats 300 LEDs so far i discovered :
inoise8mover-only one side lights
easing-doesn't go even half a strip
dotbeat-light on both sides but nothing in the middle
blur -only one side lights
most of effects kinda slugish.. animations speed is fastest

can you come up with solution?

encor50, 2017-10-15:
update ! i added 9 more effects codes are on the same links 
https://pastebin.com/RMPasRAU
https://pastebin.com/ZLmpYK87

sisensee, 2017-10-16:
Man this stuff makes sense to me one minute and loses me the next! I thought Hey it should be simple to change the candy cane to blue (matching my Christmas tree)... ive spent the last 30 minutes reading and getting more lost!

encor50, 2017-10-16:
Guys im not a programmer i also dont know what i am doing :) i get lost all the time but adding these codes was easy its just add lines to this code from other codes some codes didnt work meany of them has lines in commen and meany i still did not yet because its not easy :)

KodiMkd, 2017-10-16:
That's fine and we don't complain or critique. We are grateful for what you are doing and just letting you know that some of the effects don't work. When/if you can ,check them out,if not -no problem. Not rush or obligation. Thank you so far i appreciate.

1337hium, 2017-10-20:
@sisensee 
to change the colors search this line:
`  setupStripedPalette( CRGB::Red, CRGB::Red, CRGB::White, CRGB::White);`
then change it to your needs
like: `setupStripedPalette( CRGB::Blue, CRGB::Blue, CRGB::White, CRGB::White);`

BABYFACEPI, 2017-12-04:
Hi everyone 
Thanks for this code.
Need somme help, I’m really really novice, I’ve done this project all things run proprelly but, I wouldlike to made an automation witch

BABYFACEPI, 2017-12-04:
Hi everyone
Thanks for this code.
Need somme help, I’m really really novice, I’ve done this project all things run proprelly but, I wouldlike to made an automation witch Will change the light effect every minute for exemple.
Can someone help me with this please...

lance36, 2018-01-02:
the code from @encor50 works. i believe it even has what @BABYFACEPI  is looking for (loop between effects)