bruhautomation/ESP-MQTT-JSON-Digital-LEDs #35: Pragma Issues

jkvalence, 2017-07-01:
Hey Ben,

So putting this together and it seems that the data is not being transmitted to the strips.  I am getting this notice in the IDE upon upload.  I can't tell what is going on except that the LEDs are not working.  

> In file included from C:\Users\jvalence\Desktop\Udemy\ledcloud\ledcloud.ino:25:0:
> 
> C:\Users\jvalence\Documents\Arduino\libraries\FastLED/FastLED.h:17:21: note: #pragma message: FastLED version 3.001.003
> 
>      pragma message "FastLED version 3.001.003"
> 
>                      ^
> 
> In file included from C:\Users\jvalence\Documents\Arduino\libraries\FastLED/FastLED.h:65:0,
> 
>                  from C:\Users\jvalence\Desktop\Udemy\ledcloud\ledcloud.ino:25:
> 
> C:\Users\jvalence\Documents\Arduino\libraries\FastLED/fastspi.h:110:23: note: #pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output
> 
>      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"

My setup is simple to test this.  I am connecting my ESP8266 to cloudmqtt to test this before putting it in HA.  I can see message publish to cloudmqtt and see messages publish to the ESP8266.  Below is a sample message from the serial monitor.

> Message arrived [strip/sub] {"state":"ON","color":{"r":50,"g":50,"b":50},"brightness":255,"effect":"solid"}
> 
> solid
> Setting LEDs:
> r: 50, g: 50, b: 50

Any ideas?

Justin

yzg011, 2017-07-20:
same issue

Makr91, 2017-07-22:
Pragma messages should not be getting marked as warnings or errors - that's a bug you should file with the IDE. In the meantime, you can disable them by including #define FASTLED_INTERNAL before #include <FastLED.h>.

federom, 2018-06-03:
Same here ! How did you solve it ?

fsanano, 2019-02-09:
@Makr91 Your solution worked for me, thx!