bruhautomation/ESP-MQTT-JSON-Digital-LEDs #40: RGBW

mpcgit, 2017-08-06:
Is there any support for RGBW

ramkithepower, 2017-08-23:
I have the exact same question... But no answer...

ramkithepower, 2017-08-24:
FastLED does not support RGBW at this time. Hence no. RGBW will not be supported.

codmpm, 2017-12-28:
Seems that RGBW is possible now: https://github.com/FastLED/FastLED/issues/482#issuecomment-351179360

Anyone of you already used RGBW, yet?

Reflectin88, 2018-01-24:
Does anyone have RGBW working. I've tried using DotNetDann/ESP8266-MQTT-JSON-SK6812RGBW-HomeAssistant fork however I can't get it to work correctly. The switch in home assistant just doesn't work correctly

pacobyte, 2018-01-24:
@Reflectin88 You should post an issue in DotNetDann's repo if that's the one you're using. My guess is you'll probably get better help there since he is more likely to respond there.

pacobyte, 2018-01-24:
@Reflectin88 I just checked and noticed that he is not allowing issues on his fork. My mistake.

How is the switch in Home Assistant not working correctly?

Reflectin88, 2018-03-03:
So when I turn the switch off it turns the strip off however the switch after a couple of seconds changes back.

Reflectin88, 2018-03-04:
@pacobyte I have tested with bruh's code and it works however the colors are obviously wrong as im using a neo pixel rgbw ring

Reflectin88, 2018-03-06:
Finally worked it out, the off packet was larger then the max that pubsub could send

kurniawan77, 2018-07-28:
Would you please share the fix? Having the same issue

Regards

Reflectin88, 2018-08-28:
@kurniawan77 I have lost the code, I'm going to have a look at doing it again as I need the code and if I get it working ill post it here.

profi2111, 2019-08-25:
Did you already redo the code? I wanted to program a controller for my sk6812 rgbw neopixel ring and don't understand how to implement the warmwhite.