bruhautomation/ESP-MQTT-JSON-Digital-LEDs #42: Light strip not lighting up.

Marauder0408, 2017-08-21:
I've got the circuit set up properly and the NodeMCU software was uploaded as well. I have the proper components and automations set up in Home Assistant and I can tell that the NodeMCU receives the proper command but the strip does not light up.

What kind of voltage/amps should I expect on the output pin? I'd like to confirm the output is working properly before I try to get a replacement strip from Amazon.

Thanks!

ramkithepower, 2017-08-23:
Is the pin number correct? In your code if the following correct and have you connected the data in to the D5 pin of the nodeMCU.

#define DATA_PIN    5

In some nodeMCU the D5 pin is not after D4, you will have 3v and GND in between D4 and D5.

timmo001, 2017-10-16:
If you take a look at the pinout for the NodeMCU, D5 is the pin after the ground and 3.3v. See here:
https://cdn3.bigcommerce.com/s-vt19phz/product_images/uploaded_images/pinout.png?t=1491966337
If this still isn't working, try another pin and change the `#define DATA_PIN 5` to the new pin in the code

federom, 2018-06-04:
I have the same issue with FAST LED, the issue is which is the DATA_PIN value i should use with D5

federom, 2018-06-05:
I solved ! If you are using NodeMCU version 3, and want to use D5 , add

#define FASTLED_ESP8266_NODEMCU_PIN_ORDER

before importing the Fast Led Library,

Then it will work when using DATA_PIN 5

Hope this helps,

zaine68, 2018-09-03:
#define FASTLED_ESP8266_NODEMCU_PIN_ORDER doesn't work for me. in fairness, nothing does... so far. I know my system works as I can use the McLighting code which works a treat. But, that code uses the NeoPixel.h library.

The mqtt side of things is working fine as i can see the messages coming in from HomeAssistant. I know FastLED works with my LEDS because i can do a simple script which turns the lights on just to prove the software works. I usually use D2 labelled on the board which effectively is GPIO4, which is also the same as i used in McLighting. LEDS are WS2812B, wouldn't have thought that the colour order would stop them from working, just give them the wrong values is all from what I've read.

Only other thing i can think of is that i've had to add the int MILLION = 100000; section to get over the error that the MILLION caused in the temp section. Should that be at the very beginning or lower down in the JSON bit?

I'm just rather puzzled!