bruhautomation/ESP-MQTT-JSON-Digital-LEDs #46: pasted the code but parseObject() failed! hint welcome!

sladaar, 2017-09-29:
First of all many thanks for sharing the code, i've been following your project for long time but i just now started to do something.
So, as in object i tried to control the nodemcu with mqtt and i got in the arduinoIDE serial monitor the "parseObject() failed" error. 
Can you help me to fix it?

Thanks in advice!

timmo001, 2017-10-16:
This is likely to do with HA sending the bad JSON string.
If you monitor what `bruh/porch/set` is receiving with a tool like MQTTLens to you might be able to solve your problem

sladaar, 2017-10-16:
many thanks @timmo001 for the reply. 
I managed to get it working only in HA directly, because i really don't know why with mqttfx and publishing just the same words as HA outputs, it keeps giving the "parseObject() failed" error.

But thanks anyway, i will probably use it as intended.

KodiMkd, 2017-10-17:
i got the same problem..i dont even know what im i doing on mqtt ,im all lost

Baloowozzy, 2018-04-30:
Same problem, can't get past "parseObject() failed" with MQTT lens or Home assisant.
Related to that, can't get Home assistant to even send a MQTT message (with the "Services" tool)
```
{"entity_id":"light.bank",
"transition":"50",
"brightness":255,
"effect":"rainbow"
}
```
dosent show up in MQTTlens at all.
HASSIO Config:
```
light:
  - platform: mqtt_json
    name: Bank
    state_topic: "bank"
    command_topic: "bank/set"
    effect: true
    effect_list:
```

With MQTTlens i can publish to "bank/set, with "parseObject() failed" as a result, so i guess the NodeMCU is OK so far.
> "Message arrived [bank/set] 
> parseObject() failed"

Any hint as to where i should start looking?

sladaar, 2018-05-01:
well, i'm sorry to say that i got it to work but now some time has passed and i don't rememer how i did it.
I think it could have to check that topic and message are correct, then a workaround or another NodeMCU code (or another esp8266, esp32, or better sonoff with tasmota which is easier) to get HA and mqtt to work. If i remember correctly the problem was with the payload, which has to be different from the one a mqtt sniffer reads. Don't remember why tho!
Sorry to be not much helpful.

Baloowozzy, 2018-05-01:
no problem, every options i worth exploring.
and you seem to be on to someting with the payload.
the nodemcu responds to this payload sent from MQTTLens:
```
  "state": "on",
  "color": {
    "r": 255,
    "g": 255,
    "b": 255
  },
  "brightness": 255,
  "effect": "solid"
```

And from what i can gather, the formatting looks like this sent from Home assistant: 
```
{
  "brightness": 255,
  "color_temp": 155,
  "color": {
    "r": 255,
    "g": 255,
    "b": 255,
    "x": 0.123,
    "y": 0.123
  },
  "effect": "colorloop",
  "state": "ON",
  "transition": 2,
  "white_value": 150
}
```
Source: [https://www.home-assistant.io/components/light.mqtt_json/]

Can this possible be the problem?

excecutor, 2019-02-03:
Hi to all,

Did anyone find a solution that error ? i face same problem , when send ON command