bruhautomation/ESP-MQTT-JSON-Digital-LEDs #48: Initial status

euf0ria, 2017-10-10:
Thank you Ben, this is awesome rocketscience (still learning as i go).

I have managed to change the initial effect to fire instead of full-on white but struggle to find a way to change the initial state to ON instead of OFF. My goal is to have the strip light up as soon as the power is switched on and the esp8266 has booted up.  
Where in the code can you change this?

encor50, 2017-10-11:
you have to use trigger ? when ur esp goes online turn on the light ?

euf0ria, 2017-10-11:
So you are implying that the strip is off when powered on just because the current state in  home assistant is off (because is goes to off when esp is pulled offline)?

I will try an automation which sets the strip state to on and use esp onlinestatus as trigger. 
But, isn't this going to rely on the device_tracker monitoring the esp? I dont want this because my tracker only checks status every 30 seconds and i intend to keep it that way.

Compare this to a meelight or yeelight smartbulb: They do not depend on home assistant and device tracking. When i give them power via a regular wall  switch they light up and then report current state to home assistant.

I want the same for this led-strip: To be able to set the initial state to on, this will then report to home assistant via MQTT with the correct state.

KodiMkd, 2017-10-12:
can you leave the code so us (little rusty) to use it ?? and also can you do random effects in the loop ?? because i think will be boring same order effects non stop. i wish i can find like 100+ effects that work with this bruh automation so more choice-the better.

euf0ria, 2017-10-12:
@encor50 
Got it sorted out! One word in the code did the trick.

Look for these lines: (note that the initial state is set to false)
/******************************** GLOBALS for fade/flash *******************************/
bool stateOn = true;        //true = Strip ON when powered on. false = strip OFF when powered on

@bruhautomation: It would be awesome if you could add this comment in your code to make is easier for newcomers to find this tweak.

closed 2017-10-12 by euf0ria