bruhautomation/ESP-MQTT-JSON-Digital-LEDs #49: sometimes works/doesnt work

KodiMkd, 2017-10-12:
i uploaded the code on esp8266 like 25 times and only 2 out of 25 it fired up. samenumbers same code.. anything bugged or what is going on?? on resetting the 8266 dos it loose the code or just resets the board?

guillaume1410, 2017-10-12:
it just resets the board. 
I use the code this week end with two wemos d1 mini and it works like a charm. Are you sure that there is only one device with the same name on the mqtt bus (I made the mistake and it took me quite a lot of time to understand).
did you try to use a mqtt client with some serial monitoring instead of home assistant to perform some troubleshooting ?

KodiMkd, 2017-10-12:
umm..how do i check mqtt bus ?and how to use mqtt client?? im not realy handy in these things,i just follow tutorials and examples/templates

guillaume1410, 2017-10-12:
First (to be sure) : did you set up an mqtt bus somewhere (perhaps a raspberry pi ?). 
Are you on mac or windows or linux ?

KodiMkd, 2017-10-12:
windows..
esp8266 is connected to mqtt  right.. ijust asdd mqtt addres and password in the code and its connecing

guillaume1410, 2017-10-12:
ok. I use mqtt.fx as the client. It's java based (don't know if you've got java on your set up). otherwise you can try to find another mqtt client.

After connect to your mqtt server using the client you've just uploaded and try to publish something to the light_set_topic in your code. something like {"state": "ON", "brightness": 192}

KodiMkd, 2017-10-12:
ok my mqtt works..this is my logic of the things that should happens: i upload the code and the esp8266 connects to mqtt via wifi and execute the code -turn of the leds..so i can manual torn on from HA (as it is in bruh video).
now this is my code :
#define FASTLED_INTERRUPT_RETRY_COUNT 0 //makes animations smoother... supposedly... 
#define FASTLED_ALLOW_INTERRUPTS 0

#include <ArduinoJson.h>
#include <ESP8266WiFi.h>
#include <PubSubClient.h>
#include "FastLED.h"
#include <ESP8266mDNS.h>
#include <WiFiUdp.h>
#include <ArduinoOTA.h>

/************ WIFI and MQTT Information (CHANGE THESE FOR YOUR SETUP) ******************/
const char* ssid = "Doma"; //type your WIFI information inside the quotes
const char* password = "1236547890";
const char* mqtt_server = "m10.cloudmqtt.com";
const char* mqtt_username = "LED";
const char* mqtt_password = "led";
const int mqtt_port = 1883;
 
and this is for the leds
#define NUM_LEDS    300
#define DATA_PIN    6
//#define CLOCK_PIN 5
#define CHIPSET     WS2811
#define COLOR_ORDER GRB

 i have 1 strip of 300 leds..and i intend to add 2 more at least

guillaume1410, 2017-10-12:
what is the light_set_topic on your set up ?

KodiMkd, 2017-10-12:
/************* MQTT TOPICS (change these topics as you wish)  **************************/
const char* light_state_topic = "bruh/porch";
const char* light_set_topic = "bruh/porch/set";

const char* on_cmd = "ON";
const char* off_cmd = "OFF";
const char* effect = "solid";
String effectString = "solid";
String oldeffectString = "solid";
 
u mean this ?

guillaume1410, 2017-10-12:
yes

oops, seems like the info you've provided about the credentials are the real one ?

KodiMkd, 2017-10-12:
yes..i can see my light here going crazy

guillaume1410, 2017-10-12:
no. It's just me playing with your led strip...

guillaume1410, 2017-10-12:
blue and red flashing ?

KodiMkd, 2017-10-12:
not yet..try

KodiMkd, 2017-10-12:
different colors ..now white solid

guillaume1410, 2017-10-12:
you'd better remove the credentials you've provided me. It's not really safe. You can edit the message...

guillaume1410, 2017-10-12:
does it perform a kind of rainbow ?

KodiMkd, 2017-10-12:
i will sure np
so what can you do to help/explain/lead this blind men
and is it possible to stat random and gor andom?

KodiMkd, 2017-10-12:
yes rainbow

guillaume1410, 2017-10-12:
ok the good news is that mqtt and esp work fine

guillaume1410, 2017-10-12:
the problem seems to be on your home assistant side

KodiMkd, 2017-10-12:
ok..so..what do i do

guillaume1410, 2017-10-12:
You'd better double check the code you've paste in your home assistant configuration

guillaume1410, 2017-10-12:
do you have a functional home assistant set up or you've just setup one to play with your led strip ?

KodiMkd, 2017-10-12:
setup just for this..altough i have some alexa stuff that i can play with

yaml config ?
here 

mqtt:
  broker: m10.cloudmqtt.com
  port: 17105
  client_id: home-assistant-1
  username: LED
  password: led

light:
  - platform: mqtt_json
    name: "BRG"
    state_topic: "bruh/porch"
    command_topic: "bruh/porch/set"
    rgb_state_topic: "bruh/setcolorpub"
    rgb_command_topic: "bruh/setcolor"
    brightness_state_topic: "bruh/setbrightnesspub"
    brightness_command_topic: "bruh/setbrightness"
    effect: true
    effect_list:
      - bpm
      - candy cane  
      - confetti  
      - cyclon rainbow  
      - dots  
      - fire  
      - glitter  
      - juggle  
      - lightning
      - noise  
      - police all  
      - police one  
      - rainbow  
      - rainbow with glitter  
      - ripple  
      - sinelon  
      - solid  
      - twinkle  
    brightness: true
    flash: true
    rgb: true
    optimistic: false
  
input_slider:
  porch_animation_speed:
    name: Porch Animation Speed
    initial: 150
    min: 1
    max: 150
    step: 10

automation: 
  - alias: "Porch Animation Speed"
    initial_state: True
    hide_entity: False
    trigger:
      - platform: state
        entity_id: input_slider.porch_animation_speed
    action:
      - service: mqtt.publish
        data_template:
          topic: "bruh/porch/set"
          payload: '{"transition":{{ trigger.to_state.state | int }}}'

guillaume1410, 2017-10-12:
port should be 1883 I guess
once again credentials are in your post !

KodiMkd, 2017-10-12:
port is now 17105 ?? why 1883?

guillaume1410, 2017-10-12:
1883 is the standard mqtt port. It's the one I was using to play with your ledstrip

KodiMkd, 2017-10-12:
but why  on mqtt in have port 17105?? or that is different ?? just for esp8266 to connect on?

KodiMkd, 2017-10-12:
ok is this u playing again or me ?couse i connected and i can change  effects

guillaume1410, 2017-10-12:
it's you

guillaume1410, 2017-10-12:
I removed the connection on my side

KodiMkd, 2017-10-12:
so lemme summarize here..17105 is the port that esp is using to connect to mqtt right ?? and 1883 is the port for HA to mqtt ?

KodiMkd, 2017-10-12:
how do i let this run in loop..and/or random?? because manual is impossible to change non stop

guillaume1410, 2017-10-12:
I don't use the cloud mqtt service that you use so hard for me to answer but : according to the code you paste at the beginning your esp is using 1883 to connect to cloud mqtt.
On my side both my esp and ha connect on my self hosted mqtt service on 1883 which is the standard port.

I don't understand your last question.

KodiMkd, 2017-10-12:
well when i turn these LEDs they just keep solid white color ...until i change them manually  they dont start on their own and loop

guillaume1410, 2017-10-12:
just click on the name of your light when it's turned on and you'll have access to the effects and color. I understand that it's working now. So have fun!

KodiMkd, 2017-10-12:
ok thanks so much and sorry to bother

guillaume1410, 2017-10-12:
you perhaps just can close the issue. Happy to help

KodiMkd, 2017-10-17:
guillaume1410    OR anyone .....can u please lead me on mqtfx how o set it up if u dont mind??i wasl looking for tuorial but couldnt find it and cant realy undersand

closed 2017-10-19 by KodiMkd