bruhautomation/ESP-MQTT-JSON-Digital-LEDs #52: Deprecated: input_slider

StefanPetter, 2017-10-18:
input_slider should be replaced with input_number in the configuration.yaml file

timmo001, 2017-10-19:
I've added this to my pull request [here](https://github.com/bruhautomation/ESP-MQTT-JSON-Digital-LEDs/pull/51)

KodiMkd, 2017-10-20:
When i replace both in .yaml in HA i get error when HA starts.  That is the only place to change input_ slider to input_ number?

Eric2XU, 2017-10-22:
Here is what I use that works: 

input_number:
  animation_speed_livingroom_wall:
    name: "Animation Speed Livingroom Wall"
    initial: 5
    min: 1
    max: 150
    step: 5

Additionally, automations changed too, so now you need to use "to" instead of "state"
automation: 
  #
  # Livingroom Wall Speed Slider
  #
  - alias: "Automation Animation Speed Livingroom Wall"
    initial_state: True
    hide_entity: False
    trigger:
      - platform: state
        entity_id: input_number.animation_speed_livingroom_wall
    action:
      - service: mqtt.publish
        data_template:
          topic: "bruh/walll/set"
          payload: '{"transition":{{ trigger.to_state.state | int }}}'

jasper321, 2017-10-31:
input_number:
  Animatiesnelheid keuken:
    name: "1Animatiesnelheid"
    initial: 5
    min: 1
    max: 150
    step: 5

  - alias: "Snelheid Keuken"
    initial_state: True
    hide_entity: False
    trigger:
      - platform: state
        entity_id: input_number.animatiesnelheid_keuken
        action:
      - service: mqtt.publish
        data_template:
        topic: "ledkeuken/ledkeuken/set"
        payload: '{"transition":{{ trigger.to_state.state | int }}}'  

IT DOESNT WORK!!! PLEASE HELP

Eric2XU, 2017-10-31:
can't be sure but I think Action needs to come back one level in indent. Action and Trigger are at the same level. BTW I HATE YAML, if I had one wish it would be to go to JSON as YAML SUCKS. Actually if I had one wish it would be to go back in into the past and do whatever would be required to make sure YAML never got created in the first place.

Here is code I use:

`  - alias: "Automation Animation Speed Playroom Wall"
    initial_state: True
    hide_entity: False
    **trigger**:
      - platform: state
        entity_id: input_number.animation_speed_playroom_wall
    **action**:
      - service: mqtt.publish
        data_template:
          topic: "bruh/wallp/set"
          payload: '{"transition":{{ trigger.to_state.state | int }}}'`

Eric2XU, 2017-10-31:
Ha... here is a screenshot. 

[image]

jasper321, 2017-10-31:
https://guides.github.com/features/mastering-markdown/2017-10-31 20:29:40 ERROR (MainThread) [homeassistant.config] Invalid config for [input_number]: [Animatiesnelheid keuken] is an invalid option for [input_number]. Check: input_number->input_number->Animatiesnelheid 
[image]
[image]

euken. (See /config/configuration.yaml, line 95). Please check the docs at https://home-assistant.io/components/input_number/
2017-10-31 20:29:40 ERROR (MainThread) [homeassistant.setup] Setup failed for input_number: Invalid config.

jasper321, 2017-10-31:
can you see what the problem is?