bruhautomation/ESP-MQTT-JSON-Digital-LEDs #53: How do you go loop

KodiMkd, 2017-10-20:
Ok im trying to find out how these strips would go to loop. When i turn them ON they just solid white and dont go to loop (they supposed to?).  The code is original,just copy and paste. Do i need to do loop via mqtt i wonder or how?

KodiMkd, 2017-10-27:
Anyone??Why noone talks when is Mqtt in question??

Joeboyc2, 2017-12-06:
Hi @KodiMkd,
what do you mean when you say loop?
are you talking about effects, such as the ones Ben shows in his video?

KodiMkd, 2017-12-06:
When i say loop i mean all the effects is bruh to go trough a loop..one then after 5 mins another the a other..all of them 15 lets say..and then start all over again same thing

Joeboyc2, 2017-12-07:
OK, so you can do that with an automation in home assistant.

it would just be a case of triggering the first effect, then setting a delay on the automation, then start the next effect, and then repeat this for all you want

KodiMkd, 2017-12-07:
If you can just lead me with the first stage in HA then i can just copy the
rest

Oliver Kostov