bruhautomation/ESP-MQTT-JSON-Digital-LEDs #54: Strip flickers towards the end

riker09, 2017-10-23:
First of all, thank you for this amazing project. It is incredible how easy it was setting everything up and watch it come to life. Your YouTube channel is a great inspiration and my head is swarming with so many new ideas for projects. :slightly_smiling_face: 

Nevertheless, since I'm a rather newbie on things like Arduino and LED strips I have encountered some issues where I'm not sure where to look for the reason behind them and I'm looking for some help. If this is the wrong place than please let me know.

My setup is as following:

- NodeMCU v3
- WS2812B 5m LED strip with 60 LEDs / m (should be 300 in total, but I have not counted them)
- MeanWell RS-150-5 power supply (130W, 5 Volts, 26 Amps)

The difference in my setup to this project is that my strip is using 5V instead of 12V, so I don't need the DC-DC converter. Also I'm not using a logic shifter to shift the data signal I get from pin D5 from 3V to 5V. Aside from that it's the same thing.

Now, when I switch on the strip with solid light everything is fine (well almost, the LEDs towards the end are noticably dimmer than at the beginning, but that should be solvable by injecting power half way through, I believe). But I do get solid colors and all LEDs light up (again, I'm not 100% sure since the strip is still rolled up on the coil it was delivered on).
Okay, solid works. But when I turn on animations, there is some visible flickering towards the end of the strip. This happens on all animations. What's more puzzling is that on animations where the light is traveling down (and/or up, like `sinelon`) the strip some LEDs don't light up on each loop.

The question I have is, what's the cause of this? A faulty LED strip, the NodeMCU or could it be a software error? Or is it as simple as the missing logic shifter?

Looking forward to your answers.

zombiehunter1, 2017-10-24:
You need to provide 5V power feeds to both ends, you are getting voltage drop across the strong

riker09, 2017-10-24:
No, I haven't tried injecting power to the other side yet (no pun intended).

I have uploaded two videos of the flickering.

https://youtu.be/1ME0uWs6-eA
https://youtu.be/9XuRzFD0oO4

Is this flickering caused by low voltage on the far end? Solid colors work flawless. I would like to know the physics behind the flickering, if anyone can explain that.

encor50, 2017-10-24:
Try my code 
https://pastebin.com/RMPasRAU
https://pastebin.com/ZLmpYK87
There is more effects and i think i know why its flickering i had the same problem with my ws2813

encor50, 2017-10-24:
Because he deleted these lines form his code 
#define FASTLED_INTERRUPT_RETRY_COUNT 0
#define FASTLED_ALLOW_INTERRUPTS 0
I re added them in mine plus i have some more effects or just add these 2 lines to the top of his code and it will be good

riker09, 2017-10-25:
Thank you @zombiehunter1 and @encor50 . I will try the ``FASTLED_ALLOW_INTERRUPTS 0` thingy tonight. From what I read at the official FastLED documentation this could indeed solve my issues. Also I will checkout the added effects. Since I'm going to mount my strip behind my flatscreen tv I'm looking for more effects anyway. Running lights (light sinelon, police, etc.) are cool if the strip is visible, but probably not behind a TV.

riker09, 2017-10-25:
Okay, I tried the two paramaters @encor50 suggested. Unfortunately no success. :disappointed: 

I did try another thing: I manually set the number of LEDs to 200 (instead of 300 which are acutally in/on the strip). Sure enough the solid effect only lights up about two thirds of the strip. Sometimes when I change color or brightness the remaining 100 LEDs change color as well. It seems to be completely random, I could not make out a pattern or reproduce at will. Just clicking around in the HA light widget atm.

Effects light sinelon or police also travel only up to the set amount of LEDs, but the LEDS at the end randomly flicker in the same colors as the effect has.

encor50, 2017-10-25:
Check your cables ? If all good connected 
And change the chipset to WS2812B

riker09, 2017-10-26:
Since I don't have much time to spent on my hobby (wife + two kids, one only seven month old) I was hoping that this has happened to somebody in the past already. I did change the chipset to WS2812B right before uploading the sketch for the first time (like this:)

```
#define CHIPSET WS2812B
```

The last thing that remains which I didn't try yet is injecting more power on the other end. Given that my power supply can deliver enough Amps can I just wire the end of the strip to the beginning? This would look like a loop then:

```
Power Supply                       LED strip
  |                                    |
  V                                    V
------+                 +----------------------------+      +--------+
     +|----o------------|(+) - - - - - - - - - - - -/      / - - - - |------------+
      |    |            |(D) - - - - - - - - - - - /      /- - - - - |            |
     -|-----------o-----|(G) - - - - - - - - - - -/      / - - - - - |------+     |
------+    |      |     +------------------------+      +------------+      |     |
           |      |                                                         |     |
           |      |                                                         |     |
           |      +---------------------------------------------------------+     |
           |                                                                      |
           +----------------------------------------------------------------------+
```

I'm not very good with electricity... This won't fry my circuits, right? 😅

zombiehunter1, 2017-10-26:
This is the correct wiring, remember it's not a current problem it's a voltage problem.  the first led will receive 5v but as we get further down the string the voltage will be less due to resistance in the wire. the ws28xx chips need a stable 5 volts to work correctly.

My first experience with this was with a string 50 10mm ws2812's .  I had the same problem  powering both ends fixed it.

riker09, 2017-10-26:
Well, this sounds promising. Going to try that tonight. Will report back here.

riker09, 2017-10-26:
I hooked up power to the other end of the strip, but unfortunately this didn't change anything. The sinelon effect for instance seems even worse than before. A lot of the LEDs don't light up when they should and some show the wront color (colors and affected LEDs change randomly on each loop).

riker09, 2017-11-01:
I've decided to attach the strip to my TV regardless of the current issues. I was hoping that the flickering would go away when the strip length reduces. Unfortunately, no. :disappointed: 

Now I have one more issue. The LEDs at the top are noticably dimmer when the brightness is set to max.

### Full brightness:
[image]

### Brightness set to 50%:
[image]

What's weird about this: I inject power to both ends of the strip and the strip is not **5m** anymore but only about **3m**, which is **179** LEDs in total. When I touch the upper corner joints the brightness changes, so naturally I assumed that the connection is too "_flaky_" (excuse my choice of words, I'm not an english native). But what I mean is that the resistance at the joints is too high and I'm dealing with some sort of voltage drop(?). So I used some wire to bridge over the joints in parallel, but the brightness didn't change at all. This occures more visibly when the color is set to _white_, but it is still noticable when using other colors like _green_ or _purple_ (of course, only when brightness is set to max).

I'm not planning to illuminate the whole room with my moodlight, so for the time I'll guess I just leave it as it is and don't set the strip to full brightness. And I wanted to share (intermediate) result with you guys, since it already looks amazing:

[image]
[image]

(Unfortunately you cannot really see the flickering when converting the videos I shot on my phone to GIF.)

jainsarthak, 2017-11-15:
I have the same problem. Powering at both ends made colors better but did not fix this. 

I tried some test code from Adafruit_NeoPixel.h, that works without any issues. So I believe either it is something to do with Fastled, probably. 

@riker09 Did you try changing the Fastled option NEO_KHZ800 to NEO_KHZ400? I'll play with options tonight and report if I get anything.

If nothing works, I''ll try to reimplement it using Adafruit_NeoPixel instead of FastLED

riker09, 2017-11-15:
@jainsarthak I was just messing around with my moodlight tonight and bought a voltage meter to see if I'm dealing with voltage drop. Matter of fact, I do. Running a second power line to one of the upper cornes significantly improves brightness of the top led strip. Not to full brightness though, but it was just an experiment to see where my brightness issues were coming from. I'm going to order some more wires and will inject power on both top corners, that should take care of that.

The flickering is still another mystery to me. I did try to use Adafruit_NeoPixel.h, but my strip would not light up at all. I tried to keep the OTA and WiFi part from Bruhs original implementation intact, since the NodeMCU is mounted behind my wall-mounted TV and thus difficult to reach via USB cable. I was able to flash it a couple of times but then it seemed I broke something. Had to unmount my TV today, commented out NeoPixel stuff and got it back to status quo.

Here's the code that I tried which didn't work out:

```c
#include <ArduinoJson.h>
#include <ESP8266WiFi.h>
#include <PubSubClient.h>
#include "FastLED.h"
#include <ESP8266mDNS.h>
#include <WiFiUdp.h>
#include <ArduinoOTA.h>
#include <Adafruit_NeoPixel.h>

/************ WIFI and MQTT Information (CHANGE THESE FOR YOUR SETUP) ******************/
const char* ssid = "My WiFi"; //type your WIFI information inside the quotes
const char* password = "My super secret wifi pasword";
const char* mqtt_server = "192.168.178.99";
const char* mqtt_username = "mqtt";
const char* mqtt_password = "hahahaha";
const int mqtt_port = 1883;

/**************************** FOR OTA **************************************************/
#define SENSORNAME "tv_moodlight" //change this to whatever you want to call your device
#define OTApassword "hahahaha" //the password you will need to enter to upload remotely via the ArduinoIDE
int OTAport = 8266;

/************* MQTT TOPICS (change these topics as you wish)  **************************/
const char* light_state_topic = "home/tv_moodlight";
const char* light_set_topic = "home/tv_moodlight/set";

const char* on_cmd = "ON";
const char* off_cmd = "OFF";
const char* effect = "solid";
String effectString = "solid";
String oldeffectString = "solid";

/****************************************FOR JSON***************************************/
const int BUFFER_SIZE = JSON_OBJECT_SIZE(10);
#define MQTT_MAX_PACKET_SIZE 512

/*********************************** FastLED Defintions ********************************/
#define NUM_LEDS    179
#define DATA_PIN    5
//#define CLOCK_PIN 5
#define CHIPSET     WS2812B
#define COLOR_ORDER GRB
#define MILLION 1000000

byte realRed = 0;
byte realGreen = 0;
byte realBlue = 0;

byte red = 255;
byte green = 255;
byte blue = 255;
byte brightness = 255;

/******************************** GLOBALS for NeoPixel *******************************/
#ifdef __AVR__
  #include <avr/power.h>
#endif

// Parameter 1 = number of pixels in strip
// Parameter 2 = Arduino pin number (most are valid)
// Parameter 3 = pixel type flags, add together as needed:
//   NEO_KHZ800  800 KHz bitstream (most NeoPixel products w/WS2812 LEDs)
//   NEO_KHZ400  400 KHz (classic 'v1' (not v2) FLORA pixels, WS2811 drivers)
[... 683 lines cut ...]
            green = 255;
        } else {
            green = tmp_green;
        }
    }
    
    // blue
    if (tmp_internal >=66) {
        blue = 255;
    } else if (tmp_internal <= 19) {
        blue = 0;
    } else {
        float tmp_blue = 138.5177312231 * log(tmp_internal - 10) - 305.0447927307;
        if (tmp_blue < 0) {
            blue = 0;
        } else if (tmp_blue > 255) {
            blue = 255;
        } else {
            blue = tmp_blue;
        }
    }
}
```

Note: I did remove the effects from the original code since the fastLED is disabled anyw. Im guessing the FastLED include interfers with NeoPixel somehow=

jainsarthak, 2017-11-15:
@riker09 your strip is not lighting up with Adafruit_NeoPixel because NeoPixel does not translate your GPIO pin to DPIN. for example I use Nodemcu and for me DATA_PIN 1 in FastLED is same as pin 5 in Adafruit_NeoPixel. Both of these send data to pin labeled "D1" on my microcontroller.

You are using pin 5, that I believe will become 14 as per this:
https://github.com/FastLED/FastLED/blob/master/platforms/esp/8266/fastpin_esp8266.h#L68

`Adafruit_NeoPixel strip = Adafruit_NeoPixel(NUM_LEDS, DATA_PIN, NEO_GRB + NEO_KHZ800);
`

jainsarthak, 2017-11-16:
OMG this fixed the flickering

#define FASTLED_ALLOW_INTERRUPTS 0 

got from https://github.com/krzychb/EspFire2012/commit/524b8baae437201a1ede71cf3fb61beaa1362a10

riker09, 2017-11-17:
I *did* have the `#define FASTLED_ALLOW_INTERRUPTS 0` thing in my configuration before, but that didn't change anything so I threw it away again. But according to [issue #60](https://github.com/bruhautomation/ESP-MQTT-JSON-Digital-LEDs/issues/60#issuecomment-344978568) I have to define this __*before*__ including the FastLED library. So this needs to go in line 4.

I will try that as soon as I find some time and will report my findings back here.

Diaoul, 2017-11-20:
I'd live to know if that actually works or not. I have some WS2811 lying around and I know I could not use them with ESP8266 directly due to timing issues with more than approx 60 LEDs some time ago.
Also, do you use 3.3V-5V level shifters? I've had some success driving them directly from 3.3V.

Philje123, 2017-11-21:
I was having terrible flickering issues/wrong colours/leds remaining on etc. I am using WS2811 12v strip with 69 LEDs. And using an ESP-01.

I found that I needed to have 12v ground attached to the 3.3v ground of the ESP. This completely fixed my issues. There is no reference to this in the Bruh instructions but found reference to it elsewhere online.

I am now just having a couple of small issues where with an effect running certain sections of the strip will flicker slightly. Will try the #define FASTLED_ALLOW_INTERRUPTS 0 suggested above and report back. Good job for Arduino OTA!

craig3050, 2017-11-21:
Similar problem, using #define FASTLED_ALLOW_INTERRUPTS 0 just before declaring the FastLED library sorted my problem too. Thanks!

Philje123, 2017-11-21:
Compiled and uploaded with #define FASTLED_ALLOW_INTERRUPTS 0. Perfect now!

Thanks guys.

Just waiting for a couple of Wemos D1s to arrive and I can sort my LED strip round the TV.

riker09, 2017-11-21:
My additional wire arrived today, so I'll probably do some soldering tonight. The OTA broke for some reason, otherwise I would've tried the `#define FASTLED_ALLOW_INTERRUPTS 0` already. But now I have to dismount my TV anyway... :)

I do not use a 3.3V -> 5V level shifter. My WS2812B is directly hooked up to PIN 5 of my NodeMCU.

@Philje123 When I initially wired everything together I also had some terrible flickering because the NodeMCU and the strip were not sharing a common ground. I found a hint about this somewhere on the internet. Good to see that I'm not the only one who ran into this issue. Maybe the documentation could use some improvements?

Philje123, 2017-11-21:
It _should_ still work without the level shifter depending on the length of cable from your NodeMCU to the strip. Since the signal only has to get to the first LED and is then repeated along the chain.

I tried without the level shifter and the common ground and it seemed to work ok. But kept the shifter in the circuit since I already had it and to make sure it worked.

As for OTA I found you have to have all the upload settings matching the device you are trying to update.

And yes I think the documentation needs to mention the common ground. I wonder why this problem is not always encountered.

Diaoul, 2017-11-21:
> I wonder why this problem is not always encountered.

Because it's common knowledge. And that's also the reason why it's not explained, maybe it was obvious for the writer and he didn't think writing that down.

As for level shifting, the datasheet says it requires 5V signal, it may work at 3.3V depending on how strong the voltage is. YMMV though and if you experience problems this is the first thing I'd investigate.

For those who tried `#define FASTLED_ALLOW_INTERRUPTS 0`: do you experience that your WiFi is flaky or sometimes doesn't respond to commands?

Philje123, 2017-11-21:
Well I wouldn't say it's common knowledge. All the wiring diagrams I've seen (including this one) does not show a common ground connection.

It also seemed strange to me to have the 12v ground tied to the 3.3v ground.

I'll keep an eye on it over the next few days but so far it hasn't affected the wifi.

craig3050, 2017-11-21:
The Wifi was fine. I have made a reasonably long strip (about 280) of WS2811 for round the edge of the house at Christmas. For that I needed to inject power at least 3 points, as I was dropping to about 2.8V in the middle. 
Without using the #define FASTLED_ALLOW_INTERRUPTS 0 , I found that the lights in the middle of the chain were flickering when using some of the effects. - most noticeably with the Rainbow effect.

riker09, 2017-11-21:
I did it. The top strip is as bright as the rest of the strip thanks to some injected power. The flickering is gone thanks to the `define` bugfix. Thanks to all the little hints I got from this thread. You guys are amazing. And since this looks like a valid workaround I will go ahead and close this issue.

closed 2017-11-21 by riker09