bruhautomation/ESP-MQTT-JSON-Digital-LEDs #55: Effect List no show

ThAnKm3Lat3R, 2017-10-27:
Effect List is not showing up in Home Assistant with the code that is provided. I can see the Input Number, Automation, and Light Boxes but no section for effects.

Makr91, 2017-10-28:
Can you show us your entire .yaml configuration

ThAnKm3Lat3R, 2017-10-29:
Sorry for the delayed response....

`homeassistant:
  # Name of the location where Home Assistant is running
  name: Home
  # Location required to calculate the time the sun rises and sets
  latitude: 43.5367
  longitude: -96.6985
  # Impacts weather/sunrise data (altitude above sea level in meters)
  elevation: 428
  # metric for Metric, imperial for Imperial
  unit_system: imperial
  # Pick yours from here: http://en.wikipedia.org/wiki/List_of_tz_database_time_zones
  time_zone: America/Chicago
  # Customization file
  customize: !include customize.yaml

# Show links to resources in log and frontend
introduction:

# Enables the frontend
frontend:

# Enables configuration UI
config:

http:
  # Uncomment this to add a password (recommended!)
  # api_password:
  # Uncomment this if you are using SSL/TLS, running in Docker container, etc.
  # base_url: example.duckdns.org:8123

# Checks for available updates
# Note: This component will send some information about your system to
# the developers to assist with development of Home Assistant.
# For more information, please see:
# https://home-assistant.io/blog/2016/10/25/explaining-the-updater/
updater:
  # Optional, allows Home Assistant developers to focus on popular components.
  # include_used_components: true

# Discover some devices automatically
discovery:

# Allows you to issue voice commands from the frontend in enabled browsers
conversation:

# Enables support for tracking state changes over time
history:

# View all events in a logbook
logbook:

# Track the sun
sun:

mqtt:
  broker: 192.168.1.177
  port: 1883
  client_id: home-assistant-1
  username: username
  password: password

# Weather prediction
sensor:
  - platform: yr

switch:
  - platform: mqtt
    name: "Outlet 1"
    state_topic: "/home/test/1/state"
    command_topic: "/home/test/1"
    available_topic: "/home/test/1/available"
    payload_on: "ON"
    payload_off: "OFF"
    optimistic: false
    qos: 1
    retain: true
    
light:
  - platform: mqtt_json
    name: "Porch Strip"
    state_topic: "bruh/porch"
    command_topic: "bruh/porch/set"
    effect: true
    effect_list:
     - bpm  
     - candy cane  
     - animations  
     - template  
     - blendwave  
     - beatwave  
     - blur  
     - confetti  
     - confetti 2  
     - confetti pal  
     - dotbeat  
     - easing  
     - every n example  
     - fill grad  
     - inoise8 mover  
     - inoise8 pal  
     - noise 16 1  
     - noise 16 2  
     - noise 16 3  
     - one sine pal  
     - palette cross fade  
     - rainbow march  
     - rainbow beat  
     - ripple pal  
     - sinelon  
     - cyclon rainbow  
     - dots  
     - fire  
     - glitter  
     - juggle  
     - lightning  
     - noise  
     - police all  
     - police one  
     - rainbow  
     - rainbow with glitter  
     - ripple  
     - sinelon  
     - solid  
     - twinkle  
    brightness: true
    flash: true
    rgb: true
    optimistic: false
    qos: 0

input_number:
  porch_animation_speed:
    name: Porch Animation Speed
    initial: 150
    min: 1
    max: 150
    step: 10

automation: 
  - alias: "Porch Animation Speed"
    initial_state: True
    hide_entity: False
    trigger:
      - platform: state
        entity_id: input_slider.porch_animation_speed
    action:
      - service: mqtt.publish
        data_template:
          topic: "bruh/porch/set"
          payload: '{"transition":{{ trigger.to_state.state | int }}}'
          
# Text to speech
tts:
  - platform: google

group: !include groups.yaml
#automation: !include automations.yaml
script: !include scripts.yaml
`

derspion318, 2017-11-04:
I too am missing the effects list drop down.

`  - platform: mqtt_json
    name: "Porch Strip"
    state_topic: "home/porch"
    command_topic: "home/porch/set"
    effect: true
    effect_list:
      - bpm
      - candy cane
      - confetti
      - cyclon rainbow
      - dots
      - fire
      - glitter
      - juggle
      - lightning
      - noise
      - police all
      - police one
      - rainbow
      - rainbow with glitter
      - ripple
      - sinelon
      - solid
      - twinkle
    brightness: true
    flash: true
    rgb: true
    optimistic: false
    qos: 0
`

[image]

[image]

Makr91, 2017-11-05:
Does the effects list appear after turning on the light?

ThAnKm3Lat3R, 2017-11-05:
It doesn’t for me.

derspion318, 2017-11-08:
It appears that once the device is online (connect to wifi and talking to HASSIO it will populate the drop down in the popup card for the light. click the light to bring up the card (2nd image above) and as long as the nodemcu is online when you turn on the light it will populate.

ThAnKm3Lat3R, 2017-11-11:
@derspion318 got my effect list to show up. Now my strip wont light up....
I took my multimeter and i have 12v going into the strip. I have 4v coming from the data pin. Is there something i'm missing?

[image]

derspion318, 2017-11-11:
Shoot us a better picture of the wires.

ThAnKm3Lat3R, 2017-11-11:
@derspion318 Hopefully this helps. Let me know if it doesn't.
[image]
[image]
[image]
[image]
[image]

encor50, 2017-11-12:
i think your just missing spaces in the yaml file

encor50, 2017-11-12:
try my codes 
https://pastebin.com/RMPasRAU
https://pastebin.com/ZLmpYK87 
i have added more effects :)

ThAnKm3Lat3R, 2017-11-12:
@encor50 I am already using your code.....
So if you can find a flaw in my copying and pasting skills in the configuration.txt let me know.

ThAnKm3Lat3R, 2017-11-21:
Turned out my light strip was bad. I got a new strip and works flawlessly. Thanks for the help everyone.

closed 2017-11-21 by ThAnKm3Lat3R