bruhautomation/ESP-MQTT-JSON-Digital-LEDs #56: OpenHAB Setup

eps000, 2017-10-28:
Has anyone gotten this setup with openhab? Im trying to set it up but not too sure how to go about it. Thanks

bkpsu, 2017-12-11:
I got the gist of it working, via a MAP file:

LED.MAP
```
1={"effect":"bpm"}
2={"effect":"noise"}
3={"effect":"fire"}
4={"effect":"rainbow"}
5={"effect":"twinkle"}
6={"effect":"glitter"}
7={"effect":"solid"}
8={"effect":"sinelon"}
9={"effect":"juggle"}
10={"effect":"confetti"}
11={"effect":"dots"}
12={"effect":"lightning"}
13={"effect":"candy cane"}
14={"effect":"cyclon rainbow"}
15={"effect":"ripple"}
ON={"state":"ON"}
OFF={"state":"OFF"}
```

then use the map file in your .items description for the power/effect items:

`
Number LEDStrip_1 "LED Strip 1 Fx" {mqtt=">[broker:home/LEDStrip_1/set:command:*:MAP(LED.map)]"}`

The more variable settings (color, transition, etc..) will need rules to build the JSON string to send.

bkpsu, 2018-02-13:
@eps000 - I've finally got my setup working with openHAB fully, including the color/transition/speed rules. I wrote up a post on the OH community forums with the how-to and sample rules/items/sitemap configs (https://community.openhab.org/t/nodemcu-mqtt-led-strip-controller-build-config-how-to-videos/39902) and created some How-to videos on YouTube for my specific setup (https://www.youtube.com/playlist?list=PLH-d6dcsARGNHlRyqTrwyiWJLNb1YpbPS). I did Fork @bruhautomation 's code to add some of my own effects, fixes for WS2812 flicker, and ability to group strips into a single MQTT topic (found that works better with my christmas light setup where I had 10 strips in all my windows and door frames).

-B.K.