bruhautomation/ESP-MQTT-JSON-Digital-LEDs #59: OTA not functioning

derspion318, 2017-11-05:
I have tried uploading via OTA after initial flash and it will not accept the password. It just keeps giving me a popup with this error in the log below:

Traceback (most recent call last):
  File "C:\**PATH TO ARDUINO**\arduino-1.8.5\portable\packages\esp8266\hardware\esp8266\2.3.0/tools/espota.py", line 36, in <module>
    import logging
  File "C:\Python27\lib\logging\__init__.py", line 26, in <module>
    import sys, os, time, cStringIO, traceback, warnings, weakref, collections
  File "C:\Python27\lib\weakref.py", line 14, in <module>
    from _weakref import (
ImportError: cannot import name _remove_dead_weakref

mydakota, 2018-05-20:
Same Here even removing ota pw same issue as yours grrrrr

Smitty357, 2018-10-25:
Has there been any update on this? Having the same problem.