bruhautomation/ESP-MQTT-JSON-Digital-LEDs #61: Lights don't work

Osprey37, 2017-11-15:
Total newbie here so forgive me for asking stupid questions. I've tried to follow each step in the process and after getting everything installed on my pi and updating my Homeassistant yaml file and uploading the sketch to my board nothing happens. Homeassistant is showing the light card, the input card and the automation card but when I turn on the light nothing happens. Not sure if it's a communication issue with the ESP, or an MQTT issue or what. Is there a way to serially monitor what homeassistant is sending out and if it's communicating with the ESP? Totally stuck. Thanks for any help or guidance.

Philje123, 2017-11-21:
Can you check the MQTT log to see if it is connecting to the MQTT broker?

If it is then it's likely to be a topic related issue.

Did you make sure to set the topic etc in the code?

It's taken me a while to get my head round how MQTT works.

In my code I have:

const char* light_state_topic = "bed/lights";
const char* light_set_topic = "bed/lights/set";

Which translates to my Home Assistant light config as:

 - platform: mqtt_json
   name: "Headboard LEDs"
   state_topic: "bed/lights"
   command_topic: "bed/lights/set"

federom, 2018-06-04:
If you open the serial monitor in the Arduino IDE, when you send requests from HA, you will see the activity. 
Check if the DATA_PIN used is correct