bruhautomation/ESP-MQTT-JSON-Digital-LEDs #63: WS2801 Support

sisensee, 2017-11-25:
Any chance anyone out there has adapted this code to work with some WS2801 4 wire LEDs? I wanted to change my Chistmas tree controller over to a NodeMCU instead of a dedicated pi... 
I see that the clock pin is commented out but changing that kind of starts a lot of problems... 
For starters I get an invalid pin specified since the clock pin isnt called out in the code as being defined for anything. Hope someone smarter than me can help!

encor50, 2017-11-25:
You just have to add this to your code 
#define CLOCK_PIN 6
And change this 
#define CHIPSET     WS2801
#define COLOR_ORDER (your color order)
FastLED.addLeds<CHIPSET, DATA_PIN,CLOCK_PIN, COLOR_ORDER>(leds, NUM_LEDS);

sisensee, 2017-11-26:
You sir are a life saver! I threw this up last night thinking I had some time this weekend to work on the christmas tree lights before next weekend. SURPRISE! The wife wanted to get a tree today! You sir made that work out!! Merry Christmas!

closed 2017-11-26 by sisensee

encor50, 2017-11-26:
No prob :p and merry christmas to u