bruhautomation/ESP-MQTT-JSON-Digital-LEDs #66: Crash (exception) when changing effects (or a while after)..

larsenglund, 2017-12-06:
Hi I get the following crash when changing effects or a while (1-30s) after changing effect:

```
Exception (9):
epc1=0x4020d155 epc2=0x00000000 epc3=0x00000000 excvaddr=0x3f0c7292 depc=0x00000000

ctx: cont 
sp: 3fff0750 end: 3fff09a0 offset: 01a0

>>>stack>>>
3fff08f0:  3fff12ac 00000000 3ffeef8c 40202460  
3fff0900:  3fffdad0 00000000 3ffef41c 402056fb  
3fff0910:  61002461 6069003f 00874700 0000b20c  
3fff0920:  8f001dab 2d790125 06376303 3609454c  
3fff0930:  42350553 00363202 2e002931 192b0020  
3fff0940:  8e84ff82 ff6ec9c0 b8ff1ae2 402028eb  
3fff0950:  3ffe90b0 ac01a8c0 402021dc 402020e0  
3fff0960:  ffffffff 3ffef68d 00000000 feefeffe  
3fff0970:  00000000 00000000 00000001 3ffef970  
3fff0980:  3fffdad0 00000000 3ffef969 4020ca60  
3fff0990:  feefeffe feefeffe 3ffef980 40100718  
<<<stack<<<
```

```
Exception 9: LoadStoreAlignmentCause: Load or store to an unaligned address
Decoding 8 results
0x4020d155: PubSubClient::connected() at C:\Users\englund\Documents\Arduino\libraries\PubSubClient\src/PubSubClient.cpp line 586
0x40202460: seconds16 at C:\Users\englund\Documents\Arduino\libraries\FastLED/controller.h line 170
:  (inlined by) CEveryNSeconds::getTime() at C:\Users\englund\Documents\Arduino\libraries\FastLED/lib8tion.h line 1060
:  (inlined by) CEveryNSeconds::getElapsed() at C:\Users\englund\Documents\Arduino\libraries\FastLED/lib8tion.h line 1060
:  (inlined by) CEveryNSeconds::ready() at C:\Users\englund\Documents\Arduino\libraries\FastLED/lib8tion.h line 1060
0x402056fb: loop at C:\Users\englund\Documents\Arduino\libraries\FastLED/controller.h line 170
0x402028eb: setup at C:\Users\englund\Documents\Arduino\libraries\FastLED/controller.h line 170
0x402021dc: _M_manager at C:\Users\englund\Documents\Arduino\libraries\FastLED/controller.h line 170
0x402020e0: _M_invoke at C:\Users\englund\Documents\Arduino\libraries\FastLED/controller.h line 170
0x4020ca60: loop_wrapper at C:\Users\englund\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_main.cpp line 56
0x40100718: cont_norm at C:\Users\englund\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/cont.S line 109
```

Has anyone else experienced this? 
I'm running from a beefy bench power supply so I know I have ample power.

DirkTas67, 2017-12-10:
Can you tell me how to get this information after crashing?

larsenglund, 2017-12-10:
I use the Arduino IDE and have the serial monitor open to capture the stack trace and then paste it into "Tools->ESP Exception Decoder" in the Arduino IDE.

Joeboyc2, 2017-12-10:
I'm having similar issues with my setup too, although mine just hangs and reaet on it own, even if im not running the lights atall, its very odd, any idea on how I can troubleshoot

larsenglund, 2017-12-10:
@Joeboyc2 Connect it to a serial monitor and post the decoded exception here.

Joeboyc2, 2017-12-10:
@larsenglund, ok, I'll grab it in the morning a post It.

Thanks for the help

larsenglund, 2017-12-11:
Here's some info on how to use the exception decoder to debug: [https://github.com/esp8266/Arduino/blob/master/doc/faq/a02-my-esp-crashes.rst](https://github.com/esp8266/Arduino/blob/master/doc/faq/a02-my-esp-crashes.rst)

Joeboyc2, 2017-12-11:
Hey @larsenglund,
I cant seem to make the node throw the exception anymore.
if i do spot it again, i will be sure to post the output here