bruhautomation/ESP-MQTT-JSON-Digital-LEDs #67: Animation speed not working

hicling, 2017-12-06:
Hi my animation speed seams to do nothing. when i watch the serial monitor nothing is sent when i move the slider and I can't see any difference in the animation speed.

my code:

input_number:
  led_balkong_golv_hastighet:
    name: Led Balkong Golv Hastighet
    initial: 5
    min: 1
    max: 150
    step: 5

automation: 
  - alias: "Led Balkong Golv Hastighet"
    initial_state: True
    hide_entity: False
    trigger:
      - platform: state
        entity_id: input_number.led_balkong_golv_hastighet
    action:
      - service: mqtt.publish
        data_template:
          topic: "bruh/porch/set"
          payload: '{"transition":{{ trigger.to_state.state | int }}}'

Joeboyc2, 2017-12-06:
Hi @hicling,

i've just set this up myself, i hate the way home assistant is so fussy about the yaml files and the way they keep changing the options.
Anyways, here is what i have got, this works for me

`- id: Led_Balkong_Golv_Hastighet
  alias: "Led Balkong Golv Hastighet"
  initial_state: True
  hide_entity: False
  trigger:
  - entity_id: input_number.led_balkong_golv_hastighet
    platform: state
  action:
  - data_template:
      topic: "bruh/porch/set"
      payload: '{"transition":{{ trigger.to_state.state | int }}}'
    service: mqtt.publish`

I have moved your settings across to this code block so you should be able to just copy and paste

Joeboyc2, 2017-12-06:
if the code block above still fails you will need to double check the spacing of the first line, then make sure the spacing of the rest of the lines is as as above

hicling, 2017-12-06:
`automation:
  - id: Led_Balkong_Golv_Hastighet
    alias: "Led Balkong Golv Hastighet"
    initial_state: True
    hide_entity: False
    trigger:
    - entity_id: input_number.led_balkong_golv_hastighet
      platform: state
    action:
    - data_template:
        topic: "bruh/porch/set"
        payload: '{"transition":{{ trigger.to_state.state | int }}}'
      service: mqtt.publish`

tried this whitout luck.. :(

Joeboyc2, 2017-12-06:
did you restart home assistant?

hicling, 2017-12-06:
ye I did

hicling, 2017-12-06:
When I serial monitor the esp I can see traffic when I use the on/off and when I switch effect but nothing when is lite the animation speed bar.

Joeboyc2, 2017-12-07:
Your automation.yaml file is likely failing to load then, do you get the "invalid config" card on the main HA page.
also have a look at the info page, to review an errors you may be getting

hicling, 2017-12-07:
I dont get the invalid config. but in the log it says "/config/configuration.yaml contains duplicate key "automation"." and its true  i have this automation and "automation: !include automations.yaml" can that be the problem?

Joeboyc2, 2017-12-07:
Yes that will be the issue.
If you take that code block above, remove the first line and then paste the rest of the lines in your automations.yaml file

Make sure the spacing is the same as your other automation

Restart hone assistant and check the log again

hicling, 2017-12-08:
Thx for the help no it works! 💃

cooljimy84, 2017-12-08:
Hmm i seem to just get
ESP4/sensornode4/set {"transition":{{ trigger.to_state.state | int }}}
on my MQTT server, rather than the input number that i set the slider to.

- alias: "LED Animation Speed"
  initial_state: True
  hide_entity: False
  trigger:
    - entity_id: input_number.led_animation_speed
      platform: state
  action:
    - service: mqtt.publish
      data:
        topic: "ESP4/sensornode4/set"
        payload: '{"transition":{{ trigger.to_state.state | int }}}'

hicling, 2017-12-09:
I input this in automation.yaml and removed it from configuration.yaml.

- id: Led_Balkong_Golv_Hastighet
  alias: "Led Balkong Golv Hastighet"
  initial_state: True
  hide_entity: False
  trigger:
  - entity_id: input_number.led_balkong_golv_hastighet
    platform: state
  action:
  - data_template:
      topic: "bruh/porch/set"
      payload: '{"transition":{{ trigger.to_state.state | int }}}'
    service: mqtt.publish

Just change Id, alias and entity_id so it’s right for you

cooljimy84, 2017-12-12:
Gosh darn it, ur post above it correct, but i needed to tweak the formatting.
so the line after -  data template 
needs two more spaces, same as the next line, but service: mqtt.publish needed the two extra removed.

hicling, 2017-12-12:
Yea I can’t understand the spacing in yaml. But glad it works

Rudertier, 2018-01-22:
Dear all,
I'm afraid I do not fully understand how this automation procedure works. I have split my automations from the rest of the config: _automation: !include automations.yaml_
In there I've put:
```
alias: "Led Strip Animation Speed"
  hide_entity: False
  initial_state: True
  trigger:
    platform: state
    entity_id: input_number.led_animation_speed
  action:
    service: mqtt.publish
    data_template:
      topic: "home-assistant/led/set"
      payload: '{"transition":{{ trigger.to_state.state | int }}}'
```
How is this different from what you have been posting? Do I need to use dashes ("-")? Is this correctly formatted? 
There is no other automation set up.  But still I get the message, that my automation is not correctly formatted. 
Thank's a lot for your help.

pacobyte, 2018-01-24:
@Rudertier The hyphens/dashes can sometimes be necessary. Try this:
```yml
alias: "Led Strip Animation Speed"
  hide_entity: False
  initial_state: True
  trigger:
    - platform: state
      entity_id: input_number.led_animation_speed
  action:
    - service: mqtt.publish
      data_template:
        topic: "home-assistant/led/set"
        payload: '{"transition":{{ trigger.to_state.state | int }}}'
```

Rudertier, 2018-01-25:
@pacobyte, that worked! Many thanks.

pacobyte, 2018-01-25:
You're welcome. YAML can be a real pain in the beginning.

Now that everyone's issues are fixed in this issue, would you mind closing it, @hicling?