bruhautomation/ESP-MQTT-JSON-Digital-LEDs #70: Led colours incorrect

scottsuine, 2018-01-01:
Hi
Have got this up and running using ws2812b leds. Effects work great and control though Hassio is perfect.
One issue. When I select Solid effect and pick a colour from the picker, Red makes lights red, Green makes them blue, and Blue makes them green and all in between colours are mixed up.
The mqtt message sent to the ESP8266 is correct, sending this manually gets red correct but blue and green are reversed. SO Hassio is sending the correct values, it is the code that is not interpreting them correctly.
I have tried changed the value in the COLOR_ORDER const from RGB, GRB, BRG, GBR but this doesn’t change anything. - Colours are still wrong. Is there some other place to fix this issue in the code.
I am using the latest Fast led code.

Thanks
Scott

Joeboyc2, 2018-01-01:
Hi Scott,

I had a similar issue, it was driving my nuts, but then I found a really simple script that helped me work out which code to use.

https://github.com/FastLED/FastLED/wiki/Rgb-calibration

It's one of the examples that ships with the fast led library.

Upload that to your controller and you'll soon know which order you need to address the leds in

scottsuine, 2018-01-01:
Thanks that sorted it. All Good now
thanks