bruhautomation/ESP-MQTT-JSON-Digital-LEDs #73: Turning off

zamzon, 2018-01-20:
Hi 
My LED strip goes off randomly, is it something that can be turnd off

pacobyte, 2018-01-24:
What do you mean by, "goes off randomly"? Do you mean it actually turns **off** randomly whenever it is on? Or turns **on** randomly when it is off?

zamzon, 2018-01-24:
Hi

When it is turned on, sometimes it goes off.
I think some of the scenes turns it off after a while, I don’t think it is all of them.

pacobyte, 2018-01-24:
When the LED strip turns off by itself, does it actually report as having been turned off in Home Assistant?

zamzon, 2018-01-24:
Yes is send a OFF signal to the MQTT broker

pacobyte, 2018-01-24:
Hmm...that's odd. Does that only happen while it's "playing" one particular animation?

If any animation triggers it, does it also happen if you leave it on solid?

nickboucher, 2018-02-01:
I am also having this issue

zamzon, 2018-02-01:
I haven’t tested all the animations because it is not a fixed time it turns off.
But I have the issue with the rainbow

I haven’t tested the solid.

balsman225, 2018-02-07:
I am having the same issue. I have not tested all the effects but it definitely happened on the “twinkle” effect. HomeAssistant shows it was on for 1 hour and 45 mins before turning itself off. I will continue to test and see if the times are consistent.

Rudertier, 2018-02-11:
I have the same issue with my strip. Seem to be an issue in all automations.

nickboucher, 2018-03-01:
Has anyone made any progress on this?

Is anyone using hardware components different than the components recommended in the YouTube video? If so and you are still experiencing problems, we can begin to rule out hardware issues and assume that there is indeed a software problem.

craig3050, 2018-03-06:
It was working fine for many months however when I upgraded the device to the now JSON based approach it never operates for more than a few minutes. 

I am using an ESP12. I use WS2811 approx 5m. I do not use a level shifter as the first LED is only 10cm away.

nickboucher, 2018-03-10:
That's interesting -- I had a suspicion that the problem was related to the FastLED issue discussed in #60 (https://github.com/bruhautomation/ESP-MQTT-JSON-Digital-LEDs/issues/60), but perhaps it is related to JSON parsing? @craig3050, have you been able to your setup working correctly again?

craig3050, 2018-03-12:
Not yet, I have downloaded a fresh copy, re-installed FastLED and the other libraries, however it is still crashing after about 5-10 mins (so a small improvement but still not consistant). 

I have now deleted the arduino IDE from the computer and re-installed from scratch but so far I've not had time to re-test. I will do so ASAP and report back.

Murasakicz, 2018-03-13:
Hi, can you have serial monitor connected when strip crash so we see reason? I using ESP-01 and have crash problem because #define FASTLED_ALLOW_INTERRUPTS 0. Reason of restart was 4 "watch dog reset".

realroywalker, 2018-07-31:
Anyone ever find a solution for this ? - I recently built a small light for my kid using this setup and it randomly turns itself off (solid colour, not effect).

encor50, 2018-07-31:
https://github.com/bruhautomation/ESP-MQTT-JSON-Digital-LEDs/issues/92

highground88, 2018-12-03:
Remove " #define FASTLED_ALLOW_INTERRUPTS 0" , that fixed my issues.