bruhautomation/ESP-MQTT-JSON-Digital-LEDs #74: HA Automation question

pcdiks, 2018-01-22:
I'm trying to use a Xiaomi Cube to change the effect when I shake the cube. To do that, I created the following automation:
`
- id: ledstrip_shake
  alias: Xiaomi Cube - Shake
  initial_state: false
  hide_entity: true
  trigger:
  - platform: event
    event_type: cube_action
    event_data:
      entity_id: binary_sensor.cube_158d0001100c4e
      action_type: shake_air
  action:
  - service: light.turn_on
    entity_id: light.led_strip
    data:
      "effect": "solid"
`

HA says that my config is valid, I can see the automation in the Automation Editor in the GUI as well. But it does not work. I think it has to do with the data part of the automation but I'm not sure. Could you give me some insights? 
The shaking part of the Xiaomi Cube is working by the way. I see the event in the logs but the action is not actived.

Rawet, 2018-01-24:
I think the problem is the quotation marks, try removeing them.
My automations works just fine.

```
service: light.turn_on
      data: 
        entity_id: light.ledstripnamehere
        brightness: 255
        effect: solid
        color_name: red
```

seamus65, 2018-01-31:
Do you jave anything that changes Initial-state: false to true?
This is basically switching off the automation.
I had that very problem recently.