bruhautomation/ESP-MQTT-JSON-Digital-LEDs #75: New light effect - help

nimdy2017, 2018-02-07:
I'd like to add a new light effect to the ESP code. I'm going to put a LED light strip under a bed so that the floor around the bed glows. I'd like to add three new effects to control which side of the bed is lit up, e.g. the strip will be stuck to three sides of the bed - effect 1 would be my side of the bed and light up LEDs 1-30, effect 2 would be my wife's side of the bed LEDs 60-90, and effect would be all three sides of the bed LEDs 1-90. Is this possible?

joostthehost, 2018-06-04:
I'm also interested in a bit more information on creating custom effects. 

The effect I want to create is a wake-up-light. Thus starting at low brightness (at red/orange) and over time transitioning towards high brightness (at yellow/orange). Whenever possible, I would like to apply this effect to part of the led strip too, just like @nimdy2017 is planning his effects.

To combine both effects, we would need to know;
- how to target specific LEDs in the strip (or a group of LEDs)
- how to set a color for these LEDs
- how to set brightness for these LEDs (and/or ON vs. OFF)

federom, 2018-06-04:
I would try to do something similar. My plan is to have 2 different led strips