bruhautomation/ESP-MQTT-JSON-Digital-LEDs #77: Node mcu crashing after few minutes after selecting effect

xBiohazardx, 2018-02-23:
Hello there,
This code is pretty awsome, but as mentioned before my node mcu crashes after a few minutes. After crashing it restarts and is available in the network (pingable), but doesn't react to the mqtt messages. After pressing the reset button everything works fine for about 5 minutes again. Thanks for your help.

pcdiks, 2018-03-01:
Happens to me too, it works fine for a couple of minutes and then it crashes and becomes not responsive. I have to powercycle it to come back online.

ddtmax, 2018-03-09:
unfortunately it happens to me too, after 2 minutes with some effects

nickboucher, 2018-03-10:
This happens to me too -- I have a suspicion that the culprit is how FastLED interacts with the architecture of the NodeMCU as discussed in #60. I will be testing in a week and will report back if any of the proposed fixes in that thread work. 

Also, for reference, I believe the issue being discussed in #73 is the same as this thread.

xBiohazardx, 2018-03-11:
It seems to be stable for me now. In the Arduino IDE are two versions of the NodeMCU. I changed from version 1.0 to 0.9 and I think this fixed it for me. Leave a command if it also helped you please.

xBiohazardx, 2018-04-05:
It seems that changing the line "FastLed.delay(1000 / TransitionTime);" in function show Leds, to just "delay(1000 / TansitionTime);" fixed this issue.

xBiohazardx, 2018-06-06:
I have got 60 of them connected. But my issue on two of my strips i got a loose connection. (I am a very good engineer xD) sry for the late answer I am closing the issue if you still need help please open another one.

closed 2018-06-06 by xBiohazardx