bruhautomation/ESP-MQTT-JSON-Digital-LEDs #79: Inconsistent colors

lexam79, 2018-03-12:
Hi all,

I'm writing here, in case someone is able to provide any useful feedback, as I'm starting to go crazy with this. So I built this project, with a WS8211 strip I got off of eBay and a Wemos D1 mini. The problem I'm seeing is that it seems as if leds get "out of sync", i.e. I turn it on (solid color) all leds are on. I turn it off, some leds stay on (at the end of the strip, maybe even switch to a different color - usually green). I turn it on, then off again, a different number of leds stay on, possibly with a different color... It's completely inconsistent. Sometimes, if I turn on/off several times, I might get all leds to turn off.

I've tried several things to sort it out, both hardware and software related, to no avail.

Can anyone shed some light as to what the problem might be?

Thank you

ncareau, 2018-03-28:
Hey,

I had the same issue. After looking on the internet, I found that the WS8211 that I used prefers the data line to be directly 5v. I removed the logic level shifter and directly connected the nodecmu to the data line of the strip (12v for the power, 5v for the data line) and it fixed the problem immediately. 

I hope it helped.

lexam79, 2018-03-29:
Hi ncareau, thank you for your response. However I must be missing something, as I don't get how removing the logic level shifter might accomplish driving the strip with 5V on the data line. Isn't the Nodemcu a 3.3v unit?

ncareau, 2018-03-29:
Well, according to this image, he used 5v. : 
https://github.com/bruhautomation/ESP-MQTT-JSON-Digital-LEDs/blob/master/ESP%20MQTT%20Digital%20LEDs%20Wiring%20Diagram.png

nodemcu can be powered by up to 20v on the regulated pin. Removing the level shifter made the strip a lot more stable, but you are right, it should be 3.3

Maybe i'm missing something but the schema is maybe invalid ?

lexam79, 2018-03-30:
I think the schematic is right, but the description "5V to 3.3V level shifter" is rather misleading. This kind of level shifter are usually two-way, but in this specific application, we are looking - I think - at the 3.3V to 5V side of it. Originally, all my attempts were driving the strip directly from the output of my Wemos D1 mini, which outputs 3.3V. Having had no luck, I waited a month for a level shifter to arrive and when it did arrive, I tested it. It made no difference whatsoever. As some point, I re-tried adding `#define FASTLED_INTERRUPT_RETRY_COUNT 1`. I had already tried that @3.3V but I gave it a shot @5V too and it seemed to work. The strip was finally stable for about a day and then it went berserk again... So essentially I gave up on this and took on banging my head against the wall instead. Less frustrating... :)

TheCellMC, 2018-04-11:
I'm having the same issue. It's totally random. I bought a new esp8266 and I try again, in case my esp was faulty.

TheCellMC, 2018-04-23:
@lexam79 I got this working! I was about to give up. Connect the end of your LED strip Data to the ground on your Wemos D1.

lexam79, 2018-04-24:
@TheCellMC so you connected the end of the DATA line of the strip to ground, right?

I'll give it a shot and report back. Thank you!

TheCellMC, 2018-04-24:
Exactly. I hope that this fixes your issue

TheCellMC, 2018-04-25:
No, I meant connect the DOUT a the very end of the strip with the GND on the controller (Wemos D1 mini in your case).

edif30, 2018-05-28:
@TheCellMC While this kind of makes sense it also seems kinda crappy that the only way to correct this issue is to ground the end of the strip.  I happen to have cabinets that run long and running a line from the d-out all the way back to the 8266 would look fugly.  I think depending on your situation you can ground it almost anywhere.  Even a piece of metal that is grounded.

encor50, 2018-07-31:
https://github.com/bruhautomation/ESP-MQTT-JSON-Digital-LEDs/issues/92

twanvanbalen, 2018-10-16:
Ground the end of the LED strip to my NodeMCU fixed my problem. Grounding to the end of the power supply didn't work.