bruhautomation/ESP-MQTT-JSON-Digital-LEDs #86: Issue compiling

IamDan77, 2018-06-13:
Ive tried re-installing Arduino IDE and updated all libraries etc but i keep getting and error when I hit verify on line 420

'StaticJsonBuffer' was not declared in this scope

Unfortunatley its well beyond my blink skills. Any ideas?

IamDan77, 2018-06-13:
/********************************** START SEND STATE*****************************************/
void sendState() {
  StaticJsonBuffer<BUFFER_SIZE> jsonBuffer;

  JsonObject& root = jsonBuffer.createObject();

  root["state"] = (stateOn) ? on_cmd : off_cmd;
  JsonObject& color = root.createNestedObject("color");
  color["r"] = red;
  color["g"] = green;
  color["b"] = blue;

  root["brightness"] = brightness;
  root["effect"] = effectString.c_str();

  char buffer[root.measureLength() + 1];
  root.printTo(buffer, sizeof(buffer));

  client.publish(light_state_topic, buffer, true);
}

Joeboyc2, 2018-06-13:
Hi,
I think that method cones from the arduino is in library.

I'd that installed correctly? If it is, it may be that you have newer version, that changes the name or fictions of that method. 
Maybe try an older version

IamDan77, 2018-06-13:
I had I quick look through the ArduinoJson docs and couldnt find any differences, but Im still quite new

IamDan77, 2018-06-13:
Arduino: 1.8.5 (Mac OS X), Board: "NodeMCU 0.9 (ESP-12 Module), 80 MHz, 4M (1M SPIFFS), v2 Lower Memory, Disabled, None, Only Sketch, 115200"

/Applications/Arduino.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/Dan/Library/Arduino15/packages -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/Dan/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/Dan/Documents/Arduino/libraries -fqbn=esp8266:esp8266:nodemcu:CpuFrequency=80,FlashSize=4M1M,LwIPVariant=v2mss536,Debug=Disabled,DebugLevel=None____,FlashErase=none,UploadSpeed=115200 -ide-version=10805 -build-path /var/folders/dg/_wmwmj555hd61w_bkl8g4k280000gn/T/arduino_build_292537 -warnings=none -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.esptool.path=/Users/Dan/Library/Arduino15/packages/esp8266/tools/esptool/0.4.13 -prefs=runtime.tools.mkspiffs.path=/Users/Dan/Library/Arduino15/packages/esp8266/tools/mkspiffs/0.2.0 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=/Users/Dan/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2 -verbose /var/folders/dg/_wmwmj555hd61w_bkl8g4k280000gn/T/arduino_modified_sketch_540461/ESP_MQTT_Digital_LEDs.ino
/Applications/Arduino.app/Contents/Java/arduino-builder -compile -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/Dan/Library/Arduino15/packages -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/Dan/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/Dan/Documents/Arduino/libraries -fqbn=esp8266:esp8266:nodemcu:CpuFrequency=80,FlashSize=4M1M,LwIPVariant=v2mss536,Debug=Disabled,DebugLevel=None____,FlashErase=none,UploadSpeed=115200 -ide-version=10805 -build-path /var/folders/dg/_wmwmj555hd61w_bkl8g4k280000gn/T/arduino_build_292537 -warnings=none -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.esptool.path=/Users/Dan/Library/Arduino15/packages/esp8266/tools/esptool/0.4.13 -prefs=runtime.tools.mkspiffs.path=/Users/Dan/Library/Arduino15/packages/esp8266/tools/mkspiffs/0.2.0 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=/Users/Dan/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2 -verbose /var/folders/dg/_wmwmj555hd61w_bkl8g4k280000gn/T/arduino_modified_sketch_540461/ESP_MQTT_Digital_LEDs.ino
Using board 'nodemcu' from platform in folder: /Users/Dan/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.1
Using core 'esp8266' from platform in folder: /Users/Dan/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.1
Detecting libraries used...
"/Users/Dan/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-I/Users/Dan/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.1/tools/sdk/include" "-I/Users/Dan/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.1/tools/sdk/lwip2/include" "-I/Users/Dan/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.1/tools/sdk/libc/xtensa-lx106-elf/include" "-I/var/folders/dg/_wmwmj555hd61w_bkl8g4k280000gn/T/arduino_build_292537/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536   -DARDUINO=10805 -DARDUINO_ESP8266_NODEMCU -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_NODEMCU"   -DESP8266 "-I/Users/Dan/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.1/cores/esp8266" "-I/Users/Dan/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.1/variants/nodemcu" "/var/folders/dg/_wmwmj555hd61w_bkl8g4k280000gn/T/arduino_build_292537/sketch/ESP_MQTT_Digital_LEDs.ino.cpp" -o "/dev/null"
"/Users/Dan/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-I/Users/Dan/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.1/tools/sdk/include" "-I/Users/Dan/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.1/tools/sdk/lwip2/include" "-I/Users/Dan/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.1/tools/sdk/libc/xtensa-lx106-elf/include" "-I/var/folders/dg/_wmwmj555hd61w_bkl8g4k280000gn/T/arduino_build_292537/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536   -DARDUINO=10805 -DARDUINO_ESP8266_NODEMCU -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_NODEMCU"   -DESP8266 "-I/Users/Dan/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.1/cores/esp8266" "-I/Users/Dan/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.1/variants/nodemcu" "-I/Users/Dan/Documents/Arduino/libraries/ArduinoJson/src" "/var/folders/dg/_wmwmj555hd61w_bkl8g4k280000gn/T/arduino_build_292537/sketch/ESP_MQTT_Digital_LEDs.ino.cpp" -o "/dev/null"
"/Users/Dan/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-I/Users/Dan/Library/Arduino15/packages/esp8266/har [... cut ...]
Using library PubSubClient at version 2.6 in folder: /Users/Dan/Documents/Arduino/libraries/PubSubClient 
Using library FastLED at version 3.1.6 in folder: /Users/Dan/Documents/Arduino/libraries/FastLED 
Using library ESP8266mDNS in folder: /Users/Dan/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.1/libraries/ESP8266mDNS (legacy)
Using library ArduinoOTA at version 1.0 in folder: /Users/Dan/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.1/libraries/ArduinoOTA 
exit status 1
'ArduinoJson::JsonBuffer' is not a template

IamDan77, 2018-06-15:
OK I have fixed the issue, it was the ArduinoJSON version as JoeBoyc pointed to.
I was using version 6.0.1-beta. Dropped back to 5.13.2 changed the MILLION value to 1000000 and it worked!

Joeboyc2, 2018-06-15:
Great stuff, pleased you got to it in the end, troubleshooting it all part
of becoming a maker :-)

Have fun and let us know if you have any other issues

Joeboyc2, 2018-07-31:
Id suggest closing this issue @IamDan77 if it is resolved now

closed 2018-07-31 by IamDan77