bruhautomation/ESP-MQTT-JSON-Digital-LEDs #88: doubt about wiring? HELP

arissaraiva, 2018-06-19:
I´m not good with this wiring thing and don´t have much knowledge of arduino and esp. I was able to turn on the esp it´s comunicating with my broker and receiving the messages. But my lights are staying on the same collors and not turning on and neither turning off. What´s the necessity of having the 3.3v and GND wired? Does it affect the transmitting of data?

Joeboyc2, 2018-06-20:
Hi arissaraiva,

Can you describe what components you are using? Have you set this up the same way as Ben did in the video.

Or are you using a different esp board

3.3v is the supply voltage for an esp01, or 12e on a breakout board.

If your using a node mcu, you power that via the vin pin(5v) or the micro usb port

The esp/nodemcu needs to have a ground connection to the lights and the external power supply you are using.

Without that you will not be able to control the lights because the circuit will be broken

Maybe a photo of your setup you help us out

arissaraiva, 2018-06-20:
Hello tks for the replying, I think that´s my problem I didn´t put the ground from the light to the board, my circuit is like this:

- power supply of 12v , live and ground to led, live and ground to step down converter 5v;
- step down converter to nodemcu vin and gnd (nodemcu is alive and conecting to mqtt); 
- data pin to level shifter and level shifter to data wire in the led;

- I miss the part of vin to 3.3v and gnd to G because I didn´t understand how to do it; How can I do it?
- I didn´t connect ground from led to nodemcu because I was afraid to fry my nodemcu, how can I do it?

My led strip is a 5m ws2811 30 leds per meter RGB, I changed the collor in the sketch to match mine and leave it with 186 leds since mine is apparently 150, but some could say all over the internet that is less than this.

My led strip after everything conected was responding to the effects but not turning off.

Apreaciate your help, I don´t have a photo because I´m not at home.

tks

federom, 2018-06-20:
arissaraiva, put the same amount of led you have in your strip (count them), otherwise it may cause issues

arissaraiva, 2018-06-20:
Thanks I´d do that, but I would also like a tip or explanation of how I could ground my nodemcu toghether with the power supply whithout frying my board, I think that´s the real problem of my configuration, since I was able to run the effects but not able to turn off my led.

Joeboyc2, 2018-06-20:
Grounding the nodemcu and the power supply should work fine, you don't really need to worry about frying the board, ground is just ground, no voltage passes through that

arissaraiva, 2018-06-20:
Thank you very much will try and post the feedback afterwards

Joeboyc2, 2018-06-20:
Great stuff, hope it works out

arissaraiva, 2018-06-20:
one last question I was forwarded to this schematics: http://www.puckstar.com/ws2811-ledstrip-controlled-via-wifi-with-home-assistant/ . I was thinking about to follow these as it seems more easier, but if my led strip is 12 volts and my nodemcu is 5volts, ground directly from the led shouldn´t be a problem right? I´m really noobie on this electric thing I´m trying to learn.

Joeboyc2, 2018-06-20:
I see no issue with that schematic, you can see that the ground from the supply (plug in the wall) is connected to one of the strips ground lines, and then the 2nd ground line is connected the the node.

If you can follow that schematic with the items you have then you should be good to go

arissaraiva, 2018-06-21:
quick question when we´re talking about ground this means neutral or V- from the suply power right?

Joeboyc2, 2018-06-21:
Yes, negative, v- Ground are all pretty much the same when work ing with low voltage dc

arissaraiva, 2018-06-21:
tks sir apreaciate your help

closed 2018-07-03 by arissaraiva