bruhautomation/ESP-MQTT-JSON-Digital-LEDs #89: New custom effect

miesjel12, 2018-06-24:
Hi all,

not sure if this is the correct place for my question, but lets hope so!

I've managed to get everything working with the provided code but I'm planning on using the strip in the garden as a way of lighting up the driveway. 

If possible, i would like to add an effect that lights op a couple of leds every few feet (like spots), and when someone enters the driveway I would like it to light up the whole strip (one at a time from front to backup and than stay on). 

I've tried learning from the code in the arduino file but could not create an effect of my own. 

Michel

miesjel12, 2018-06-25:
Thanks for responding! 

But I can't controle each individual led via home assistent, i can however tell it to start with a specific effect. Or am I wrong? 

That's why I said, I need to create a new effect.

federom, 2018-06-25:
i would create an effect in the .ino file and then i would call it from an HA automation

Joeboyc2, 2018-06-25:
That's the point miesjel12 is making, he/she wants help in making the effect

miesjel12, 2018-06-26:
Exactly, thanks 😁