bruhautomation/ESP-MQTT-JSON-Digital-LEDs #90: Not responding sometimes

arissaraiva, 2018-07-03:
Hello everyone, sometimes when I leave too much time on, when I try to turned off my leds dont respond to hassio commands, it apears on my mosquitto and my network but I push the button in hassio and there's no answer, so I need to turn on and turn off several times untill it goes off. Mine leds are least than 5m from my router inside a paper box, Im using hassio in rpi with same config.yaml from the instructions, my led are connected to a nodemcu with a power supply of 12v and no level shifter attached. There's anyway I see which commands my nodemcu is receiving or not over the air?

Tks apreciate your help

bacumings, 2018-07-03:
I have this same issue sometimes. I'm not sure how to troubleshoot?

arissaraiva, 2018-07-06:
Anyone could help? Just for more info im using a  supply power of 12v 5a, doesnt seem to be the problem, and when I turn off from the energy and take him from the box I made and poke a bit it works again, really strange, maybe my nodemcu is sleeping after sometime?

encor50, 2018-07-31:
https://github.com/bruhautomation/ESP-MQTT-JSON-Digital-LEDs/issues/92