bruhautomation/ESP-MQTT-JSON-Digital-LEDs #91: issue Arduino sketch

lucavanthiel, 2018-07-19:
i have a issue with the Arduino sketch,
at "/**************************** START STRIPLED PALETTE *****************************************/"

Arduino: 1.8.5 (Mac OS X), Board:"NodeMCU 1.0 (ESP-12E Module), 80 MHz, 4M (1M SPIFFS), v2 Lower Memory, Disabled, None, Only Sketch, 115200"

sketch________1__ledvoor:972: error: variable or field 'setupStripedPalette' declared void
 void setupStripedPalette( CRGB A, CRGB AB, CRGB B, CRGB BA) {
                           ^
sketch________1__ledvoor:972: error: 'CRGB' was not declared in this scope
sketch________1__ledvoor:972: error: 'CRGB' was not declared in this scope
 void setupStripedPalette( CRGB A, CRGB AB, CRGB B, CRGB BA) {
                                   ^
sketch________1__ledvoor:972: error: 'CRGB' was not declared in this scope
 void setupStripedPalette( CRGB A, CRGB AB, CRGB B, CRGB BA) {
                                            ^
sketch________1__ledvoor:972: error: 'CRGB' was not declared in this scope
 void setupStripedPalette( CRGB A, CRGB AB, CRGB B, CRGB BA) {
                                                    ^
sketch________1__ledvoor:1032: error: variable or field 'addGlitter' declared void
 void addGlitter( fract8 chanceOfGlitter)
                  ^
sketch________1__ledvoor:1032: error: 'fract8' was not declared in this scope
sketch________1__ledvoor:1042: error: variable or field 'addGlitterColor' declared void
 void addGlitterColor( fract8 chanceOfGlitter, int red, int green, int blue)
                       ^
sketch________1__ledvoor:1042: error: 'fract8' was not declared in this scope
sketch________1__ledvoor:1042: error: expected primary-expression before 'int'
 void addGlitterColor( fract8 chanceOfGlitter, int red, int green, int blue)
                                               ^
sketch________1__ledvoor:1042: error: expected primary-expression before 'int'
 void addGlitterColor( fract8 chanceOfGlitter, int red, int green, int blue)
                                                        ^
sketch________1__ledvoor:1042: error: expected primary-expression before 'int'
 void addGlitterColor( fract8 chanceOfGlitter, int red, int green, int blue)
                                                                   ^
In file included from /Users/luca/Documents/Arduino/sketch________1__ledvoor/sketch________1__ledvoor.ino:36:0:
/Users/luca/Documents/Arduino/libraries/FastLED/FastLED.h:17:21: note: #pragma message: FastLED version 3.001.007
 #    pragma message "FastLED version 3.001.007"
                     ^
In file included from /Users/luca/Documents/Arduino/libraries/FastLED/FastLED.h:68:0,
                 from /Users/luca/Documents/Arduino/sketch________1__ledvoor/sketch________1__ledvoor.ino:36:
/Users/luca/Documents/Arduino/libraries/FastLED/fastspi.h:110:23: note: #pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output
 #      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"
                       ^
/Users/luca/Documents/Arduino/sketch________1__ledvoor/sketch________1__ledvoor.ino: In function 'void setup()':
sketch________1__ledvoor:190: error: redefinition of 'void setup()'
 void setup() {
      ^
sketch________1__ledvoor:1: error: 'void setup()' previously defined here
 void setup() {
      ^
sketch________1__ledvoor:194: error: 'setupStripedPalette' was not declared in this scope
   setupStripedPalette( CRGB::Red, CRGB::Red, CRGB::White, CRGB::White); //for CANDY CANE
                                                                      ^
/Users/luca/Documents/Arduino/sketch________1__ledvoor/sketch________1__ledvoor.ino: In function 'bool processJson(char*)':
sketch________1__ledvoor:319: error: 'StaticJsonBuffer' was not declared in this scope
   StaticJsonBuffer<BUFFER_SIZE> jsonBuffer;
   ^
/Users/luca/Documents/Arduino/sketch________1__ledvoor/sketch________1__ledvoor.ino:319:3: note: suggested alternative:
In file included from /Users/luca/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/StaticJsonDocument.hpp:8:0,
                 from /Users/luca/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson.hpp:10,
                 from /Users/luca/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson.h:9,
                 from /Users/luca/Documents/Arduino/sketch________1__ledvoor/sketch________1__ledvoor.ino:32:
/Users/luca/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/Memory/StaticJsonBuffer.hpp:110:7: note:   'ArduinoJson::Internals::StaticJsonBuffer'
 class StaticJsonBuffer : public Internals::StaticJsonBufferBase {
       ^
sketch________1__ledvoor:319: error: 'jsonBuffer' was not declared in this scope
   StaticJsonBuffer<BUFFER_SIZE> jsonBuffer;
                                 ^
sketch________1__ledvoor:323: error: 'class ArduinoJson::JsonObject' has no member named 'success'
   if (!root.success()) {
             ^
sketch________1__ledvoor:398: error: 'MILLION' was not declared in this scope
       unsigned int kelvin  = MILLION / color_temp;
                              ^
/Users/luca/Documents/Arduino/sketch________1__ledvoor/sketch________1__ledvoor.ino: In function 'void sendState()':
sketch________1__ledvoor:430: error: 'StaticJsonBuffer' was not declared in this scope
   StaticJsonBuffer<BUFFER_SIZE> jsonBuffer;
   ^
/Users/luca/Documents/Arduino/sketch________1__ledvoor/sketch________1__ledvoor.ino:430:3: note: suggested alternative:
In file included from /Users/luca/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/StaticJsonDocument.hpp:8:0,
                 from /Users/luca/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson.hpp:10,
                 from /Users/luca/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson.h:9,
                 from /Users/luca/Documents/Arduino/sketch________1__ledvoor/sketch________1__ledvoor.ino:32:
/Users/luca/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/Memory/StaticJsonBuffer.hpp:110:7: note:   'ArduinoJson::Internals::StaticJsonBuffer'
 class StaticJsonBuffer : public Internals::StaticJsonBufferBase {
       ^
sketch________1__ledvoor:430: error: 'jsonBuffer' was not declared in this scope
   StaticJsonBuffer<BUFFER_SIZE> jsonBuffer;
                                 ^
sketch________1__ledvoor:435: error: invalid initialization of non-const reference of type 'ArduinoJson::JsonObject&' from an rvalue of type 'ArduinoJson::JsonObject'
   JsonObject& color = root.createNestedObject("color");
                                                      ^
sketch________1__ledvoor:444: error: 'class ArduinoJson::JsonObject' has no member named 'measureLength'
   char buffer[root.measureLength() + 1];
                    ^
sketch________1__ledvoor:445: error: 'class ArduinoJson::JsonObject' has no member named 'printTo'
   root.printTo(buffer, sizeof(buffer));
        ^
sketch________1__ledvoor:445: error: 'buffer' was not declared in this scope
   root.printTo(buffer, sizeof(buffer));
                ^
/Users/luca/Documents/Arduino/sketch________1__ledvoor/sketch________1__ledvoor.ino: In function 'void loop()':
sketch________1__ledvoor:497: error: redefinition of 'void loop()'
 void loop() {
      ^
sketch________1__ledvoor:6: error: 'void loop()' previously defined here
 void loop() {
      ^
sketch________1__ledvoor:619: error: 'addGlitterColor' was not declared in this scope
     addGlitterColor(80, realRed, realGreen, realBlue);
                                                     ^
sketch________1__ledvoor:729: error: 'addGlitter' was not declared in this scope
     addGlitter(80);
                  ^
exit status 1
variable or field 'setupStripedPalette' declared void

 -------------------------------------------------------

please help, does anyone know something?

Joeboyc2, 2018-07-31:
hey, are you using the arduinoIDE to do the upload or something else, like platformio?

lucavanthiel, 2018-08-01:
The arduino ide

Joeboyc2, 2018-08-16:
can you upload you full sketch, looking at the errors above it says the void setup() has been redeclared

maybe if you copy and pasted you did it twice in the same sketch.....

Might be worth just downloading the original sketch again and attempting to upload that, it wont work for your setup, but at least you will know if the sketch is the issue, or your modifications

zuidwijk, 2018-09-26:
I think it has to do with the arduino or libraries... I've got the same kind of error message:
error: 'StaticJsonBuffer' was not declared in this scope
StaticJsonBuffer<BUFFER_SIZE> jsonBuffer;

The code always worked and didn't change. 

Even the default code gives this error, without any adjustments...

zuidwijk, 2018-09-26:
It seems to be a known issue... https://github.com/bblanchon/ArduinoJson/issues/756

I rolled back to ArduinoJson version 5.13.1 and everything works fine again