bruhautomation/ESP-MQTT-JSON-Digital-LEDs #96: Cyclon rainbow crashing MQTT

Necrolust, 2018-08-22:
So I have an issue where ONLY the cyclon rainbow effect crashes MQTT and I can see the device trying to reconnect (actually it says its connected over and over) to the broker but its completely unresponsive. Also says its setting the LEDs to 0,0,0 when its stuck on the cyclon rainbow

This is on a NodeMCU 1.0

Anyone else seen this?

stefanlaheij, 2018-09-26:
Yes, same issue here. As long as I don't use this effect everything keeps working perfect.

Necrolust, 2018-09-26:
I hate to plug others work but maybe worth also checking out
https://github.com/toblum/McLighting?

My memory is foggy but I either a) had a similar issue with McLighting and
fixed it or b) I fixed the crashing issue on Bruh's code

I think my issue is that my data pin was defined incorrectly or my ESP
didn't like "D4" or whatever, i changed it to straight numerical value like
"12" off the top of my head and that fixed the crashing

thainglo1, 2018-10-07:
I am having this exact issue.  Just started playing around with a strip of lights and NodeMCU today in preparation for Halloween.  Every other effect works great, just the Cyclon Rainbow somehow keeps the NodeMCU from shutting down.

Per Necrolust comment, I checked my Data Pin, it is defined as 5.

Anyone have any pointers?

radeonorama, 2018-12-23:
Hey! So i've had the same issue and just found a solution, well for me anyway. 

I've been using the 5m strip of 5050 lights, 60 LEDs/Meter but they only have a chip every 3rd led. 

At first i didn't think anything of it and with 10 meters assumed I had to set my NUM_LEDS to 600 as that's how many LEDs I have however as the chip is every 3rd, I noticed 3 leds have the same colour for each set so technically I have 200 addressable sections. 

As soon as i changed my sketch to 200 led's it all works perfectly. 

might be worth you guys checking the same! hope it helps someone.

radeonorama, 2018-12-27:
nope, why'd you ask? you been having the issue?

Martinvdm, 2019-08-22:
Same issue with Cyclon Rainbow, crashes MQTT

dajomas, 2019-08-23:
Which MQTT server ar you using? I have a 250 led WS2812b strip that is working without a problem. Never caused my MQTT server (mosquitto) to crash and the led strip just lights up as I would expect it.

Could you give some more details on your setup? How can you tell MQTT crashes? If you reset the MQTT server can you send messages for the led strip again?

Martinvdm, 2019-08-23:
I am using mosquito. It is not the MQTT server itself which is crashing but the led strip does not respond any more on MQTT commands. I have 240 leds ws2815 defined as ws2811. Tried the Pinkyleds project and did not had any issues with cyclone rainbow.

SquareBeard, 2019-09-02:
I have the same problem and the "Lightning" effect is also affected but less.
I believe it is because those two effects use a delay within a loop. Somehow the mqtt client miss the messages.

SquareBeard, 2019-09-07:
I replaced the code for the Cyclon Rainbow effect with the one from the PinkyLEDs project and it is now working fine.