bruhautomation/ESP-MQTT-JSON-Digital-LEDs #98: Animation speed not working

marrobHD, 2018-09-23:
The animation speed function is not working. The Arduino didn´t get the message. 
`- id: '1529740274996'
  alias: "Led Strip Animation Speed"
  hide_entity: False
  initial_state: True
  trigger:
    - platform: state
      entity_id: input_number.porch_animation_speed
  action:
    - service: mqtt.publish
      data_template:
        topic: "set/porch/set"
        payload: '{"transition":{{ trigger.to_state.state | int }}}'`

SeriousMikeDE, 2018-12-26:
Hi
Would you mind sharing your whole configuration.yaml file?

j2242, 2019-03-03:
I'm having the same issue with my whole setup:

light:
  - platform: mqtt
    schema: json
    name: "Porch Strip"
    state_topic: "bruh/porch"
    command_topic: "bruh/porch/set"
    effect: true
    effect_list:
      - Christmas RGB Loop
      - Christmas Sparkles
      - bpm
      - USA
      - candy cane
      - confetti
      - cyclon rainbow
      - dots
      - fire
      - glitter
      - juggle
      - lightning
      - noise
      - police all
      - police one
      - rainbow
      - rainbow with glitter
      - ripple
      - sinelon
      - solid
      - twinkle
    brightness: true
    flash: true
    rgb: true
    optimistic: false
    qos: 0
    
  - platform: mqtt
    name: "Test Strip"
    command_topic: "bruh/mqttstripTEST/setpower"
    state_topic: "bruh/mqttstripTEST/setpowerpub"
    rgb_state_topic: "bruh/mqttstripTEST/setcolorpub"
    rgb_command_topic: "bruh/mqttstripTEST/setcolor"
    brightness_state_topic: "bruh/mqttstripTEST/setbrightnesspub"
    brightness_command_topic: "bruh/mqttstripTEST/setbrightness"
    animationspeed_command_topic: "bruh/mqttstripTEST/setanimationspeed" 
    optimistic: false
    
  - platform: mqtt
    name: "Bedroom Window Strip"
    command_topic: "bruh/mqttstripBRwindow/setpower"
    state_topic: "bruh/mqttstripBRwindow/setpowerpub"
    rgb_state_topic: "bruh/mqttstripBRwindow/setcolorpub"
    rgb_command_topic: "bruh/mqttstripBRwindow/setcolor"
    brightness_state_topic: "bruh/mqttstripBRwindow/setbrightnesspub"
    brightness_command_topic: "bruh/mqttstripBRwindow/setbrightness"
    animationspeed_command_topic: "bruh/mqttstripBRwindow/setanimationspeed"
    optimistic: false
    
    
input_select:
  test_led_effect:
    name: Test LED Effect
    options:
     - "Christmas"
     - "Christmas Sparkles"
     - "Christmas Static RGB"
     - "Candy Cane"
     - "Holly Jolly"
     - "Valentine"
     - "Lovey Day"
     - "St Patty"
     - "Easter"
     - "USA"
     - "Independence"
     - "Go Blue"
     - "Hail"
     - "Touchdown"
     - "Halloween"
     - "Punkin"
     - "Thanksgiving"
     - "Turkey Day"
     - "BPM"
     - "Confetti"
     - "Cyclon Rainbow"
     - "Dots"
     - "Fire"
     - "Glitter"
     - "Juggle"
     - "Lightning"
     - "Police All"
     - "Police One"
     - "Rainbow"
     - "Rainbow with Glitter"
     - "Ripple"
     - "Sinelon"
     - "Solid"
     - "Twinkle"
    initial: "Solid"
    
  brwindow_led_effect:
    name: Bedroom Window Effect
    options:
     - "Christmas"
     - "Christmas Sparkles"
     - "Christmas Static RGB"
     - "Candy Cane"
     - "Holly Jolly"
     - "Valentine"
     - "Lovey Day"
     - "St Patty"
     - "Easter"
     - "USA"
     - "Independence"
     - "Go Blue"
     - "Hail"
     - "Touchdown"
     - "Halloween"
     - "Punkin"
     - "Thanksgiving"
     - "Turkey Day"
     - "BPM"
     - "Confetti"
     - "Cyclon Rainbow"
     - "Dots"
     - "Fire"
     - "Glitter"
     - "Juggle"
     - "Lightning"
     - "Police All"
     - "Police One"
     - "Rainbow"
     - "Rainbow with Glitter"
     - "Ripple"
     - "Sinelon"
     - "Solid"
     - "Twinkle"
    initial: "Solid"

input_number:
  porch_animation_speed:
    name: Porch Animation Speed
    initial: 150
    min: 1
    max: 150
    step: 10
    
  test_animation_speed:
    name: Test Animation Speed
    initial: 150
    min: 1
    max: 150
    step: 10
    
  brwindow_animation_speed:
    name: Bedroom Window Animation Speed
    initial: 150
    min: 1
    max: 150
    step: 10
    

automation:
  - alias: "Porch Animation Speed"
    initial_state: True
    hide_entity: False
    trigger:
      - platform: state
        entity_id: input_slider.porch_animation_speed
    action:
      - service: mqtt.publish
        data_template:
          topic: "bruh/porch/set"
          payload: '{{ trigger.to_state.state | int }}'
          
  - alias: "Test Input Effect"
    initial_state: True
    hide_entity: True
    trigger:
      - platform: state
        entity_id: input_select.test_led_effect
    action:
      - service: mqtt.publish
        data_template:
          topic: "bruh/mqttstripTEST/seteffect"
          payload: '{{ trigger.to_state.state | string }}'
          
  - alias: "Test Animation Speed"
    initial_state: True
    hide_entity: False
    trigger:
      - platform: state
        entity_id: input_slider.test_animation_speed
    action:
      - service: mqtt.publish
        data_template:
          topic: "bruh/mqttstripTEST/setanimationspeed"
          payload: '{{ trigger.to_state.state | int }}'
          
  - alias: "Bedroom Window Animation Speed"
    initial_state: True
    hide_entity: False
    trigger:
      - platform: state
        entity_id: input_slider.brwindow_animation_speed
    action:
      - service: mqtt.publish
        data_template:
          topic: "bruh/mqttstripBRwindow/setanimationspeed"
          payload: '{{ trigger.to_state.state | int }}'
          
  - alias: "Bedroom Window Input Effect"
    initial_state: True
    hide_entity: True
    trigger:
      - platform: state
        entity_id: input_select.brwindow_led_effect
    action:
      - service: mqtt.publish
        data_template:
          topic: "bruh/mqttstripBRwindow/seteffect"
          payload: '{{ trigger.to_state.state | string }}'
          
          `

j2242, 2019-03-03:
After pasting here I finally noticed the issue:
When the MQTT system updated a while back input_slider was renamed as input_number which was a breaking change. Most of us got that fixed but failed to update the subsequent automation which called on this topic because it did not create an error. 
It fixed mine, hopefully anyone else with the same issue will see this and catch it.