bruhautomation/ESP-MQTT-JSON-Digital-LEDs #102: inconsistent colors

miesjel12, 2018-11-04:
Hi all,

I've a strange problem with my led strip and this script. Everything is working but when I pick a mixed color like yellow for example, the strips starts with a green kind of color and ends with red. With the requested yellow color in the middle. It does this with every color example the green, red and blue ones. 

How can i fix this?

Its a 15 meter strip, with power injected each several meters. It contains 720 leds, i've entered this number in the config in the line NUM_LEDS. 

I'm using a nodeMCU.

JefersonSchramm, 2019-02-12:
Same problem here with 300 leds. I can't set solid color, except blue, red and green. All the other color start with one color and finish with other. Did you fixed it?

JefersonSchramm, 2019-02-13:
I've fixed the problem. The problem is not on the code but in the power injection. I have 5m and injecting power in the middle and the end of strip solved the problem.