bruhautomation/ESP-MQTT-JSON-Digital-LEDs #103: Compiling error 'StaticJsonBuffer' was not declared in this scope

jccarey5, 2018-11-22:
When compiling i get an error:

exit status 1
'StaticJsonBuffer' was not declared in this scope

This is the location of the error
/********************************** START SEND STATE*****************************************/
void sendState() {
  StaticJsonBuffer<BUFFER_SIZE> jsonBuffer;

  JsonObject& root = jsonBuffer.createObject();

  root["state"] = (stateOn) ? on_cmd : off_cmd;
  JsonObject& color = root.createNestedObject("color");
  color["r"] = red;
  color["g"] = green;
  color["b"] = blue;

  root["brightness"] = brightness;
  root["effect"] = effectString.c_str();

  char buffer[root.measureLength() + 1];
  root.printTo(buffer, sizeof(buffer));

  client.publish(light_state_topic, buffer, true);
}

Is there a fix for this?

Joeboyc2, 2018-11-22:
Use the older version of arduinojson, v5x

andilge, 2019-10-19:
according to ArduinoJson, Migrating from version 5 to 6 on
https://arduinojson.org/v6/doc/upgrade/

StaticJsonBuffer has to be changed to StaticJsonDocument and
DynamicJsonBuffer to DynamicJsonDocument

please go to the scetch in Arduin ID, find 
cJsonBuffer
and replace
cJsonDocument

It'll find and replace 2 codes. Save and pre compile to test. Worked fine for me and now I'm at the newest ArduinoJson lib version