bruhautomation/ESP-MQTT-JSON-Digital-LEDs #104: LEDs will not light up with this script. MQTT works, Circuit tested with another script and works

hazoomeh, 2018-11-24:
Spent a few days trying to get this to work. I'm having strange issues with this, managed to get to the stage where the HA showing the switch, colour selection and effects. Problem is LED's aren't switching on at all. I've checked my data pin and it's correct. My MQTT works and shows the requests from HA in Ardunio IDE serial monitor. I flashed my nodemcu with the neopixel strandtest code, as a test and the LED's work. Not entirely sure what is going on. Any help is much appreciated! Thanks﻿

SevenEvens, 2019-03-22:
Hi Hazoomeh, I had the same issue and did the same tests. But the issue was that I thought I bought WS2811 and I bought WS2812 so after I change the chipset to WS2812 it works perfect, maybe you are facing same issue :)

ufo56, 2019-03-30:
WS2812 or WS2811 wich one is correct ?

SevenEvens, 2019-04-01:
For me the correct one was WS2812